antiprism/mpd_oled #3: Display flicker and frame size issue

vjkolaventy, 2018-06-18:
Firstly thank you for the nice work and the detailed installation. I have tried it on RPI Zero. The display was flickering. The upper part of the screen has some overlay of date and other data which I am not sure. Please see attached photo and video.

[image]

[vid.zip](https://github.com/antiprism/mpd_oled/files/2112440/vid.zip)

Regards
Vijay

antiprism, 2018-06-19:
Hi Vjay

Thanks for reporting the issue.

I am not immediately sure what the problem is, but I wonder if it relates to the transmission of the data not being good enough at the speed the i2c bus is being run, and this leading to screen corruption.

It looks like you are running with the same set-up as me (I have an RPI Zero W with a 4 pin i2c SH1106 OLED) and I was able to run with the bus speed set to 1200000, with the OLED connected by 15cm cables (although I currently use 10cm cables). My OLED screen can be connected to 3.3V or 5V, and I have it connected to 5V.

If you are using 20cm cables, you could try using shorter cables. If you are running the i2c bus faster than 400000, you could try running it at 400000 (dtparam=i2c_arm_baudrate=400000 in /boot/config.txt, and run mpd_oled with '-f 15'). If you are using 3.3V, and your screen supports 5V, try using that.

Adrian.

vjkolaventy, 2018-07-16:
Dear Adrian

I have been trying various settings and recently I have tried using your code with RPI 3. I do not have the issued of flicker with RPI 3 and I get clean display initially. But as the music is played, the frame slowly moves upwards. I am using the following command

mpd_oled -o 3 -a 3c -b 10 -g 1 -f 25

Please see attached video and help me.

[video.zip](https://github.com/antiprism/mpd_oled/files/2196873/video.zip)

Regards
Vijay

vjkolaventy, 2018-07-16:
Dear Adrian

Just to update you that your code is working fine with Moode. This problem is only with Volumio.

Regards
Vijay

antiprism, 2018-07-16:
Hi Vijay

My understanding is that the SSD1306 OLED display you are using has a hardware scrolling function, and this wraps the display. What you are seeing looks like this kind of scrolling, as the mpd_oled program writes to absolute coordinate locations and these have been vertically shifted on the screen and wrapped.

The code I use to access the OLED includes scrolling functions, but I don't call any of them.

You may be seeing the same problem as here,

   https://forums.adafruit.com/viewtopic.php?f=47&t=60867

but with the difference that because mpd_oled is not using scrolling the scroll offset is never reset.

The scroll may be an occasional error from running your particular hardware setup too fast (you are using mpd_oled -f 25 and your wires look at least 15cm long). Kernel differences between Volumio and Moode may also be a factor. Do you still see the scroll in Volumio with dtparam=i2c_arm_baudrate=400000 and mpd_oled -f 15?

Anyway, I may be able to clear the issue for you by periodically resetting the scroll. I'll look at this.

Adrian.

antiprism, 2019-11-17:
Closing. I haven't received any other reports of this issue and so haven't implemented  the scroll reset. I will reopen the issue if there are new reports.

antiprism, 2021-01-18:
New report of what seems to be the same issue

https://community.volumio.org/t/oled-spectrum-display-on-volumio-on-raspberry-pi-zero-w/8067/224

antiprism, 2021-04-07:
Closed with https://github.com/antiprism/mpd_oled/commit/bb559271b351e0129ac5f4ad8159c76dbcfeaa4b

closed 2021-04-07 by antiprism