antiprism/mpd_oled #4: Spotify an Airplay

wynniphu, 2018-06-20:
Hello dear developer,
man thanks for your code 👍 
Have you planed to integrate Spotify and Airplay?
thanks

wynniphu

antiprism, 2018-06-20:
Hi wynniphu

When I looked at Spotify on Volumio, it required a premium account to access the service, which I don't have. I don't know what is involved with supporting Spotify and cannot test anything, so I am not really well placed to add it in.

I have previously been told that Airplay is not working, but I had no further information and I don't have any way to test this. However, I was able to stream DLNA to the player (Volumio, I think), and the spectrum display was working fine with the audio, but the stream did not include track information and so this information could not be displayed on the OLED.

I'll leave the issue open.

Adrian.

wynniphu, 2018-06-20:
Hey Adrian,

thank you for your quick answer. Currently there is on offer on Spotify. 3 months for 0,99€.
If you have time, it would be great, when you can implement this nice feature.

antiprism, 2018-06-20:
Thanks for letting me know about the offer.

antiprism, 2018-06-21:
Thanks for the update. When the mpd_oled program is running on Volumio it gets the information it can from Volumio directly, and other information from MPD. MPD is used for a copy of the audio stream for the spectrum analysis, and also for the audio bitrate value. As these two things are missing on the display it sounds like the Volumio Spotify plugin is not using MPD to play the audio. I'll ask about this on the Volumio forum.

What is displayed on the OLED when playing through Airplay?

antiprism, 2018-06-21:
Ok, I can see in the Volumio forum that the Spotify does not play through MPD

   https://volumio.org/forum/spotify-connect-t2811-30.html

It may be possible to work around this, for the spectrum analyser, by configuring ALSA to make a copy of the audio stream. However, the CAVA author was unable to make this work on an RPi, and it seems possible that the changes could interfere with Volumio

   https://github.com/karlstav/cava#from-alsa-loopback-device-tricky

antiprism, 2018-06-21:
I am not sure if it is possible to turn off power to the display through software, but I will look at if it is possible to blank the screen when mpd_oled is stopped.

antiprism, 2018-06-22:
I have just pushed a change to blank the screen when mpd_oled terminates.

It looks like it may be possible for the Volumio Spotify plugin to provide a copy of the audio data for the spectrum analyser. I have asked about this in the Volumio forum, but have not yet receieved a reply

   https://volumio.org/forum/spotify-connect-t2811-380.html#p49993

wynniphu, 2018-06-22:
Thanks for your work! I will test on monday.

wynniphu, 2018-07-02:
Hey Adrian,
today I tested my Volumio player with Airplay. There is no spectrum data.

antiprism, 2018-07-03:
Hi wynniphu

Thanks for the feedback on Airplay.

I haven't received any replies on how I could reconfigure the Volumio Spotify plugin. However, when I reviewed the software (librespot) that the plugin is based on in more detail, I believe that it only produces one output stream. In which case, configuring a FIFO would mean no audible output, which doesn't help!
And, as Airplay is also affected, it would be better to find a general solution for sources that do not use MPD rather than try to fix them individually.

I have previously looked at an ALSA solution, and have had some success using a different technique to the one mentioned by the Cava project. I have been able to duplicate the audio that is playing to a file and write it in the right format. However, if this file is a FIFO and mpd_oled is not running then the audio stops playing (blocks) as soon as the FIFO buffer is full, which is almost instantly. A solution to this is an intermediate program that is always running and reads the ALSA output FIFO and writes it to the mpd_oled input FIFO, and if the mpd_oled input  FIFO is not being read the intermediate program would discard the ALSA data, and so ALSA would never stop if mpd_oled was not running.

I think this technique can work, and will likely mean spectrum data for all output. However, making it work means customising the ALSA config file, which is autogenerated by Volumio, and so requires patching the Volumio source code appropriately. This might be difficult, as the extra configuration is not just a fixed block of text but has to rewrite what Volumio would be writing in a different way.

It may be possible to achieve the same result with a program that writes the FIFO by interfacing directly with ALSA. This should then avoid having to patch Volumio and will also avoid the blocking issue. Likewise, it should also work on other players without patching. This seems a much better approach, so I will look into this next (I have already seen some code that might be suitable as a base). 

While it is not obvious, the multiroom/snapcast plugin issue mentioned in the following Volumio forum thread is exactly the same problem, as the snapcast plugin also depends on a FIFO written to by MPD, and so doesn't work with the Spotify plugin

   https://volumio.org/forum/multiroom-and-synced-multiroom-should-built-t8264.html

Adrian.

wynniphu, 2018-07-04:
Thanks for your detail answer. 
In the meantime, it is possible to hide the spectrum for AirPlay and Spotify?

antiprism, 2018-07-04:
Hi wynniphu

There is currently no way to hide the spectrum, but I am making progress with the ALSA solution.

Adrian.

supercrab, 2018-08-10:
If you have an USB audio device with mic in you **can** configure CAVA to use this as the audio source and not the MPD FIFO.

Find these lines in main.cpp

`"method = fifo\n"`
`"source = /tmp/mpd_oled_fifo\n"`

And change to 

`"method = alsa\n"`
`"source = <INISERT> \n"`

I found the name of the audio source by using `aplay -l`.  In my case I used `hw:5,0`

**This works like a charm with all plugins!**   BUT you have to split the audio output cable and route it back to the mic input

antiprism, 2018-09-18:
Hi Marcus

Sorry for the delay following this up.

I have just pushed changes to allow the Cava input method and source to be specified on the command line. You should now be able to use  'mpd_oled -c alsa,hw:5,0'.

My local software-only solution continues to work well. I have included the unfinished FIFO server program in the repository (audio_multi_fifo.cpp), but for reference only. I want to look at other software solutions before completing it or producing instructions, as it may not be needed.

Adrian.

[P.S. I also incorporated your change to only show the kbitrate if the value was valid]

supercrab, 2018-09-19:
Hi Adrian!

Thanks for the update!

Will test later 👍🏼

Cheers

antiprism, 2018-09-20:
I have been able to produce a working /etc/asound.conf using the ALSA loopback device (based on the Cava example asound.conf file).

Load the module
`sudo modprobe snd-aloop`

I am using an i2s soundcard with the softvolume control on Volumio. Adapting the generated configuration file
```
pcm.softvolume {
    type             plug
    #slave.pcm       "softvol"
    slave.pcm       "softvolume2"
}

pcm.softvol {
    type            softvol
    slave {
        pcm         "plughw:1,0"
    }
    control {
        name        "SoftMaster"
        card        1
        device      0
    }
max_dB 0.0
min_dB -50.0
resolution 100
}

pcm.softvolume2 {
    type plug
    slave.pcm {
        type multi
        slaves {
            a { channels 2 pcm "softvol" }  # the real device
            b { channels 2 pcm "hw:Loopback,0" }  # the loopback driver
        }
        bindings {
            0 { slave a channel 0 }
            1 { slave a channel 1 }
            2 { slave b channel 0 }
            3 { slave b channel 1 }
        }
    }
    ttable [
        [ 1 0 1 0 ]   # left  -> a.left,  b.left
        [ 0 1 0 1 ]   # right -> a.right, b.right
    ]
}
```
I then run `mpd_oled -c alsa,hw:Loopback,1` and have a working spectrum for MPD, Spotify, or whatever is playing.

The sample rate is 48000, but it looks like Cava is handling the downsampling to 44100 for ALSA input (it doesn't for FIFO input) and so it doesn't need to be done in the asound,conf file.

The next problem is to come up with a set of instructions for the general case...

Maybe there are only a limited number of generated configuration files for each player system, and so a replacement could be provided for each, and then installed in a way that it can't be modified by future reconfiguration.

Svaras, 2019-02-03:
Dear Adrian,

want to thank you for very nice plugin you have made.

I also get stuck with problem of the Spotify connect  spectrum visualizing on Volumio. The same thing with song time progress bar (It's only moves when I forward song manually).

As I understand using ALSA is note the best solution because of downsampling rate. 

Have you any plans or solutions to get this done with FIFO?

antiprism, 2019-02-03:
Hi Svaras

Thanks for the feedback.

The resamplimng shouldn't be a problem here because it happens in Cava (which calculates the spectrum) after the stream is split, i.e. the audio stream you hear isn't resampled, just the audio data that is analysed for the spectrum. I did write a FIFO server but I couldn't avoid the need for a custom ALSA configuration file, and in the end it seemed easier to just do everythything in the ALSA configuration file.

For the progress bar, does it match what you see in the web UI? If it does, then I probably can't fix it. If it doesn't then could you get a command line session on the Volumio box while using Spotify and, while there is a progress mismatch, run 'volumio status" twice with a a short delay between runs (say 3 seconds, just to see how things are changing) and give an estimate of the progress percentage in the UI and on mpd_oled, and post this along with the output of the 'volumio status' commands.

Adrian.

Svaras, 2019-02-03:
Hi Adrian,

You was right regarding song progress bar. Web UI and 'volumio status' both ways I do not get elapsed song time information correctly. It should be spotifyconnect plugin problem. But it's not a big deal to live without this progress bar :) 

So the ALSA option should be the best way then. Will it work with any sound card with Volumio software Volume control option on?

antiprism, 2019-02-04:
Hi Svaras

The ALSA configuration file I posted is for an I2S DAC with software volume control enabled.

I can't currently provide support for ALSA configuration, because I don't know enough about it and because the player systems are each writing a configuration file that depends on the player and the hardware and also other software like equalizers are setting up a custom configuration.

In your case, the file I posted above may work with minimal changes.
Run 'aplay -L' to get the device name. In my case, using software conversions 

```
plughw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dac, 
    Hardware device with all software conversions
```

This gives me a device name of plughw:sndrpihifiberry,0. Whatever your device name is, substitute it where I have plughw:1,0 in the example file, and follow the other instructions I gave in my earlier post.

Adrian.

sburavtsov, 2019-09-27:
I was able to configure Volumio/mpd_oled to playback via hardware mixer of my DAC and capture stream to loopback device.
It is kind of hack - eg I edited tmpl files and removed some templates of Volumio integrations, but it works.
There is two points:
1. Create default hw device to playback data (dac) and loopback device and duplicate output to it
2. Configure each output plugin to use only default output and default mixer (here is hack as each plugin uses own config options, but basically it is simple)

See my config files attached.

/etc/asound.conf:
(replace hw:IQaudIODAC with hw:<YOURDACCARDNAME> - aplay -l output)
```
pcm.!default {
  type asym
  playback.pcm "LoopAndReal"
  capture.pcm "looprec"
}

ctl.!default {
	type hw
	card IQaudIODAC
	device 0
}

pcm.looprec {
    type hw
    card "Loopback"
    device 0
    subdevice 0
}

pcm.LoopAndReal {
  type plug
  slave.pcm mdev
  route_policy "duplicate"
}

pcm.mdev {
  type multi
  slaves.a.pcm pcm.i2sdac_pcm
  slaves.a.channels 2
  slaves.b.pcm pcm.MixLoopback
  slaves.b.channels 2
  bindings.0.slave a
  bindings.0.channel 0
  bindings.1.slave a
  bindings.1.channel 1
  bindings.2.slave b
  bindings.2.channel 0
  bindings.3.slave b
  bindings.3.channel 1
}

pcm.MixLoopback {
  type dmix
  ipc_key 1025
  slave {
    pcm "hw:Loopback,0,0"
    rate 48000
    periods 128
    period_time 0
    period_size 1024 # must be power of 2
    buffer_size 8192
  }
}

pcm.i2sdac_pcm {
	type plug
	slave.pcm "hw:IQaudIODAC,0"
}
```
[root.zip](https://github.com/antiprism/mpd_oled/files/3664552/root.zip)

antiprism, 2019-09-28:
Hi Sergey

Well done on getting this working, and thanks for sharing your configuration!

Adrian.

antiprism, 2019-12-18:
For reference, here is a solution that is working for me on Moode, but might need tweaking. The MPD output is written to hw:0,0 and so the following configuration overrides the "hw" plugin and duplicates the first two channels to the loopback device.

Load the loopback module
`sudo modprobe snd-aloop`

This is /etc/asound.conf
```
pcm.!hw {
        @args [ CARD DEV SUBDEV ]
        @args.CARD {
                type string
                default {
                        @func getenv
                        vars [
                                ALSA_PCM_CARD
                                ALSA_CARD
                        ]
                        default {
                                @func refer
                                name defaults.pcm.card
                        }
                }
        }
        @args.DEV {
                type integer
                default {
                        @func igetenv
                        vars [
                                ALSA_PCM_DEVICE
                        ]
                        default {
                                @func refer
                                name defaults.pcm.device
                        }
                }
        }
        @args.SUBDEV {
                type integer
                default {
                        @func refer
                        name defaults.pcm.subdevice
                }
        }

        type plug
        slave.pcm {
                type multi
                slaves {
                        a { channels 2 pcm {
                                type hw
                                card $CARD
                                device $DEV
                                subdevice $SUBDEV
                        }
                        }  # the real device

                        b { channels 2 pcm "plughw:Loopback,0" }  # loopback driver
                }
                bindings {
                        0 { slave a channel 0 }
                        1 { slave a channel 1 }
                        2 { slave b channel 0 }
                        3 { slave b channel 1 }
                }
        }
        ttable [
                [ 1 0 1 0 ]   # left  -> a.left,  b.left
                [ 0 1 0 1 ]   # right -> a.right, b.right
        ]
        hint {
                show {
                        @func refer
                        name defaults.namehint.extended
                }
                description "Direct hardware device without any conversions"
        }
}
```
Start mpd_oled with
`sudo ./mpd_oled -o 6 -b 21 -g 1 -f 25 -c alsa,plughw:Loopback,1`

Adrian.

fabio323ti, 2020-04-13:
Script is working for loopback but no more audio output .  Moode config also have empty list of selectable output device. 
just try to remove asound.conf and restart but .. isn't change, loopback still active but MPD have no output.

antiprism, 2020-04-14:
> Script is working for loopback but no more audio output . Moode config also have empty list of selectable output device.

For reference, replied to on the Moode forum: http://moodeaudio.org/forum/showthread.php?tid=155&pid=18586#pid18586

fabio323ti, 2020-04-19:
Finally script is tested and working on moode  using DAC board, but only for default output in MPD.conf.
If we set to use GraphicEq or ParametricEq output, cava stop working, like missing loopback.
Maybe need some adjust in asound.conf?
Thanks Adrian.

antiprism, 2020-04-19:
Has your /etc/asound.conf file been overwritten? If so, post the new file. If not, post the output of
```mpc outputs```

fabio323ti, 2020-04-19:
No, asound.conf still the same as your last post code (with pcm.!hw) , don't touch it.
unless Eqgraph or ParametricEq was selected all was working, and cava loopback was ok. 
after selecting one of them eq, mpe output change of course to "alsaeq" or "eqfa4p".. and no more loopback capture on Loopback:1.

antiprism, 2020-04-19:
If you look at /etc/alsa/conf.d/alsaequal.conf, it looks like this
```
pcm.alsaequal {
    type plug
    slave.pcm "plug_alsaequal";
}

ctl.alsaequal {
    type equal
    controls "/usr/local/bin/alsaequal.bin";
}

pcm.plug_alsaequal {
    type equal
 	slave.pcm "plughw:0,0";
    controls "/usr/local/bin/alsaequal.bin";
}
```
That means that it is outputing to 'plughw' rather than 'hw', which is why I guess the asound.conf file isn't duplicating the audio. I would think the best solution is to amend your asound.conf file to "sandwich" a duplication plugin between alsaequal and alsaequal_plug, following the method used in https://github.com/antiprism/mpd_oled/issues/4#issuecomment-423255641

supercrab, 2020-11-05:
Hi

I tried using your config for loopback for volumio but it didn't work me.  :( I can hear the audio but there's no visualisations.  If I set the playback output in volumio to Loopback then that works, so loopback device is OK but the audio isn't duplicated.

Is there anything else I can try?

antiprism, 2020-11-05:
Hi Mase

I don't have Volumio up and running at the moment.

Are you using Software volume control? I think that this setting caused Volumio to generate a base /etc/asound.conf, which I then modified by sandwiching in a plugin to make the audio copy (the file I posted).

If you are not using software volume control, maybe you can modify Sergey's file https://github.com/antiprism/mpd_oled/issues/4#issuecomment-536098398 , replaing his "hw" output device with yours.

Otherwise, if Volumio did not create a an /etc/asound.conf file on your system, you might be able to use the file I created for Moode, which will create an audio copy for anything written to any "hw" device (use "plughw:Loopback,1" to write to the loopback device!)

Adrian.

supercrab, 2020-11-05:
Hi Adrian!

Yeah, I'm using software audio control and I changed the /etc/asound.conf that was generated as per your example.

I will try Sergey's file and your Moode file.  It's frustrating because ALSA is quite complicated and pretty sure I'm close to getting it working.

I'll give Sergey's file a try

antiprism, 2020-11-05:
Hi Mase

In that case, if you have the original file, perhaps you could post that, and also your modified version. I don't think Sergey's file, or my Moode file, will work for you (if Volumio is outputting to an AlSA plugin called "softvolume").

Adrian

supercrab, 2020-11-25:
UPDATE: I've thrown the Pi Zero in the bin and have gone with a Pi 3.  I can get 50 FPS and the display is so much more responsive!  The mpd_oled and CAVA processes don't hog the CPU and the system is infinitely more snappier.  

I manged to get the software loopback working fine (as per your asound.conf example) but unfortunately the Spotify Connect2 plugin has the occasional annoying stutter.  This happens even when CAVA and mpd_oled are not running - I've seen other people have a similar issue.  There is no stuttering with the hardware mixer though but alas I've not been able to get loopback working with hardware.

antiprism, 2021-02-26:
Hi

Here is an `/etc/asound.conf` file for Moode which intercepts the ALSA configuration for the graphic equalisers on Moode 7. It doesn't a reference to the sound card. I think this might work sufficiently well that I would support it, in which case I will write a script to install and uninstall it

```
pcm.!eqfa12p {
    type copy
    slave.pcm "eqfa12p_and_copy"
}

pcm.eqfa12p_and_copy {
    type plug
    slave.pcm {
        type multi
        slaves {
            a { channels 2 pcm "plug_eqfa12p" }       # the original output
            b { channels 2 pcm "plughw:Loopback,0" }  # the loopback driver
        }
        bindings {
            0 { slave a channel 0 }
            1 { slave a channel 1 }
            2 { slave b channel 0 }
            3 { slave b channel 1 }
        }
    }
    ttable [
        [ 1 0 1 0 ]   # left  -> a.left,  b.left
        [ 0 1 0 1 ]   # right -> a.right, b.right
    ]
}
pcm.!alsaequal {
    type copy
    slave.pcm "alsaequal_and_copy"
}

pcm.alsaequal_and_copy {
    type plug
    slave.pcm {
        type multi
        slaves {
            a { channels 2 pcm "plug_alsaequal" }     # the original output
            b { channels 2 pcm "plughw:Loopback,0" }  # the loopback driver
        }
        bindings {
            0 { slave a channel 0 }
            1 { slave a channel 1 }
            2 { slave b channel 0 }
            3 { slave b channel 1 }
        }
    }
    ttable [
        [ 1 0 1 0 ]   # left  -> a.left,  b.left
        [ 0 1 0 1 ]   # right -> a.right, b.right
    ]
}
```
In `/etc/modules` I have the following line to load the looppback device
```
snd-aloop
```
The Loopback device loads as first sound card for me at boot, which would mean no playback ausio, and so I force it to be not first by creating the file `/etc/modprobe.d/alsa-base.conf` and adding the line
```
options snd-aloop index=-2
```
Reboot and **enable either graphic equaliser**, then run mpd_oled with a command like
```
mpd_oled -o6 -c alsa,hw:Loopback,1
```

Adrian.

supercrab, 2021-02-26:
Hi

I’ll give it a test for you. It looks similar to the configuration I had that worked but it’s hard to tell for sure. I find these ALSA config files harder to work with than C++ or JavaScript  No wonder why you can’t support them.  It would be good to knock it on the head

alain93, 2021-04-21:
Hi Adrian,
 I used the solution for Moode and it worked fine, do you have a solution for Volumio?

antiprism, 2021-04-22:
Hi Alain

I don't have a specific st of instructions, but there are two example configurations earlier in the thread

   https://github.com/antiprism/mpd_oled/issues/4#issuecomment-423255641
   https://github.com/antiprism/mpd_oled/issues/4#issuecomment-536098398

My understanding is that I will be able to provide a reliable solution for Volumio 3 (when it is released).

Adrian.

alain93, 2021-04-22:
ok, thanks Adrian

supercrab, 2021-04-22:
@antiprism when Volumio 3 comes out hopefully we'll be able to get the ALSA thing working properly.  I tried a version of my plugin that Balbuze had modified on the beta Volumio 3, but it wouldn't work for me.
Hopefully I'll be able to sort this out...

fabio323ti, 2021-07-22:
@sburavtsov , your setup works like a charm with volumio, just volume change not affect any output..
it is working in your setup too?
thanks

antiprism, 2022-11-05:
Closing. The loopback deviice and ALSA configuration can now be enabled in the browser UI on Moode (loopback setting) and Volumio 3 (mpd_oled plugin in store, currently in beta).

closed 2022-11-05 by antiprism