antiprism/mpd_oled #5: Which font  is base for your Display

Roughtrade, 2018-06-23:
I would have the following question: which tff do you use as the basis for your excellent code. I have a script that let me IR remote control the current IP address show us sysinfos. For this I stop via a script the mpd_oled service and show 10 seconds the info. Thereafter, the mpd_oled service is restarted. That works quite well so far, but unfortunately the fonts do not match yours.
Many thanks in advance

Peter

antiprism, 2018-06-23:
Hi Peter

I am using the Adafruit_GFX library and its default screen font, which is a 7x5 bitmap font (displayed in a 8x6 box). The font is included as C code

   https://github.com/adafruit/Adafruit-GFX-Library/blob/master/glcdfont.c

The larger text on the clock display is the same font, but scaled by an integer.

The following ttf font looks pretty close (I don't know if it has all the same characters or if it scales the same way)

   https://www.dafont.com/code-7x5.font

Adrian.

Roughtrade, 2018-06-25:
thanks, for this tip.

closed 2018-06-25 by Roughtrade