antiprism/mpd_oled #8: try to compile error

Roughtrade, 2018-06-27:
PLAYER=MOODE make
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMOODE  -c -o main.o main.cpp
main.cpp:28:25: fatal error: programopts.h: Datei oder Verzeichnis nicht gefunden
 #include "programopts.h"
                         ^
compilation terminated.
<eingebaut>: die Regel für Ziel „main.o“ scheiterte
make: *** [main.o] Fehler 1

regards Peter

antiprism, 2018-06-27:
Hi Peter

Thanks for reporting this. There were some new files with the last commit that I didn't add to the repository. I have now added these, and checked that the Github code builds.

Adrian.

P.S. I have added the IP address to the clock screen, but I want to test on other players before pushing the changes.

Roughtrade, 2018-06-28:
Hi Adrian,
Thanks for the update with the scrolling text. The compiling works now.
I look forward to your in the postscripum announced extension to the IP address.
I already have a small workaround where I stop the mpd_oled daemon and start my python script. I will implement my idea to the playlists and the song selection similarly. This is exactly what I imagined.

http://codelectron.com/rotary-encoder-with-raspberry-pi/

I'm not overriding that with a rotary-encoder but with lirrcc.

Regards Peter

antiprism, 2018-07-15:
I pushed the change (around two weeks ago) to include the IP address and connection type/strength on the clock screen.

closed 2018-07-15 by antiprism