antiprism/mpd_oled #10: Switch between 12hr and 24hr format

elcuevero, 2018-07-16:
Is there a way to switch between 12hr and 24hr format? Tried changing the code in display.cpp, line 79 to  
strftime(str, STR_SZ, "%l:%M:%P", now);
and reran install.sh but there where no changes. This is in Volumio.

antiprism, 2018-07-17:
Hi

You will also need to re-run 'make' to build the program (but with the position and size of the main clock screen the am/pm will not be visible without other changes).

However, regarding my reply on the Volumio forum, I have already added in a time format option locally (for 24h/12/h, with/without leading zero) and I will probably push these changes later today.

Adrian.

elcuevero, 2018-07-17:
That's perfect, Thank you very much!!

antiprism, 2018-07-17:
I have pushed the changes for this. 'mpd_oled -C 3' specifies a 12h clock with no leading zeros.

Adrian.