antiprism/mpd_oled #12: show the IP adress of the device

stefangrotz, 2018-11-23:
Hey, first of all thanks for the great work. 
For a lot of use cases in a home network it would be very handy if one could read the ip adress of the device on the screen. It could be shown very small in one of the corners.

I will fork this project later and work on this. Any tipps how to do this?

antiprism, 2018-11-23:
Hi

The program does currently try to show the IP address on the "clock" screen, which is displayed when nothing is playing. The following video shows the IP address being detected and displayed when the player starts up: https://youtu.be/Jl63Dt5A0Jo

Adrian.

stefangrotz, 2018-11-23:
Ah thats great, thanks for the quick answer.

closed 2018-11-23 by stefangrotz