antiprism/mpd_oled #14: SH1106 SPI support

naude-r, 2018-12-10:
hi,

have tried getting mpd_oled to run with an sh1106 spi. seems like more work is required to make it work correctly. to sanity check have used luma.oled's example apps. this does work, so the wiring and other setup is correct.

have checked that the basics (RESET, DC, chip select) is correct. have also updated oled_is_spi_proto accordingly.

do you have any specific idea on what needs to update?

antiprism, 2018-12-10:
Hi

Thanks for the feedback. I have just reviewed the OLED library code, and it does not directly support SH1106 with SPI. If I have understood, you are running with "mpd_oled -o 6" and have modified oled_is_spi_proto  to call the SPI setup for OLED_SH1106_I2C_128x64. However, it looks like ArduiPi_OLED::display(void)  in ArduiPI_OLED.cpp will also need changes as it is calling fastI2Cwrite(buff, 17); for OLED_SH1106_I2C_128x64, which will be wrong for SPI.

Adrian.

antiprism, 2018-12-10:
Also, but I think you probably reviewed this, some of the the SPI parameters cannot be set on the command line and have to be set in init_display() in display.cpp if the defaults need changing.

naude-r, 2018-12-11:
hi,

thank you for the response. i did review the SPI parameters. there were all fine except for the default chip select value.

will review ArduiPi_OLED::display(void).

naude-r, 2018-12-11:
have modified ArduiPi_OLED::display(void) as follows (when isSPI is true):
>     bcm2835_gpio_write(dc, HIGH);
>     if (oled_type == OLED_SH1106_SPI_128x64)
>     {
>       fprintf(stdout, "is SSH1106 spi\n");
>         for (uint8_t k=0; k<8; k++)
>         {
>           sendCommand(0xB0+k);//set page addressSSD_Data_Mode;
>           sendCommand(0x02) ;//set lower column address
>           sendCommand(0x10) ;//set higher column address
>         }
>     }
> 
>     // Send all data to OLED
>     for ( i=0; i<oled_buff_size; i++)
>     {
>       fastSPIwrite(*p++);
>     }

still no joy. have to admit i have no idea what i am doing ;-)

antiprism, 2018-12-12:
All I can suggest is that in the original code 
`sendCommand(0xB0+k);//set page addressSSD_Data_Mode;`
was called eight times, with different values of k, each time followed by writing some data. The way you have it all the data is written at the end (outside the k loop).

naude-r, 2018-12-12:
hi,

got the display to switch on and display something. unfortunately just random noise / snow.

had to upgrade to the latest bcm2835 (v1.58) to make it work. the current version, as on trunk, caused an endless loop in bcm2835_spi_transfer.

have you seen the random noise issue before? guess that something is still off in ArduiPi_OLED::display(void).

ArduiPi_OLED::display(void) has now been updated to mirror the i2c:
 ```
    bcm2835_gpio_write(dc, HIGH);
    if (oled_type == OLED_SH1106_SPI_128x64)
    {
      char buff[17] ;
      uint8_t x ;
      buff[0] = SSD_Data_Mode;
      for (uint8_t k=0; k<8; k++)
      {
        sendCommand(0xB0+k);//set page addressSSD_Data_Mode;
        sendCommand(0x02) ;//set lower column address
        sendCommand(0x10) ;//set higher column address

        for( i=0; i<8; i++)
        {
          for (x=1; x<=16; x++)
            buff[x] = *p++;

          fastSPIwrite(buff, 17);
        }
      }
    }
```
do you perhaps have any other advice?

antiprism, 2018-12-12:
It looks like with the fastI2Cwrite(buff, sz) call the first character is a "command", rather than data, and so 17 bytes are sent for 16 bytes of data. This does not appear to be the case for fastSPIwrite(buf, sz), in which case you could try fastSPIwrite(&buff[1], 16); in your code above.

I updated bcm2835 to v1.25 in a development version which uses autotools for the build (I think because I encountered a compilation issue). If possible, I'll update this to v1.58. I haven't had any feedback of success or failure with the supported SPI OLED (mpd_oled -o 1) and therefore haven't had confirmation that the version of bcm2835 in master works correctly with it.

naude-r, 2018-12-12:
thank you for the quick response.
unfortunately the change made no difference:
```
fastSPIwrite(&buff[1], 16);
```
suspect something else is wrong. the display is already corrupted after ArduiPi_OLED::begin (with a 10s pause).

antiprism, 2018-12-13:
It's a shame it didn't work out. Just a thought, if it is suitable for your project, have you checked whether your display also supports I2C?

naude-r, 2018-12-13:
yes, it does, but that is not the default. to change to I2C one would have to resolder resistors or some such....unfortunately out of my league.

naude-r, 2018-12-13:
found a bug. have not tested the fix yet.

the issue is with the sequence:
1. set DC pin high to indicate data
2. send various commands, e.g. SH1106_Set_Page_Address, which pulls DC low for command node.
3. by the time we do fastSPIwrite dc is actually low.

have corrected the above by adding a sendData overload that takes a char* and length. will test later today.

naude-r, 2018-12-13:
the above mentioned bug addressed the issue. all working now.

thank you for your the help!

antiprism, 2018-12-14:
That is great. Well done for finding the problem!

I'll leave the issue open for reference.

Theojong, 2019-05-07:
hello i have a question i have an oled display ssd1325 spi but i can't get mpd_oled spi to work can you help me with that already thanks

Theojong, 2019-05-07:
I am a layman in this area if I send my SSD1325 to spi then iksudo ./mpd_oled-o 1 -r 24 -b 10 -g 1 -f 20 that does not want you can help me with this my ssd1325 display send to me with raspberry pi 3 hifiberry please help me

bedankt theo

antiprism, 2019-05-07:
Hi Theo

mpd_oled only supports displays compatible with SH1106 or SSD1306, and from a quick web search it appears that SSD1325 is not compatible with either of these, and so your display will not work with mpd_oled. I am not able to add support for more displays at this time, but it is something I would like to look at in the future.

Adrian.

Theojong, 2019-05-07:
okay thanks well i will now get a new ssd1306 i2c display i hope if it goes better then

greeting Theo.

antiprism, 2020-02-27:
Closed by: https://github.com/antiprism/mpd_oled/commit/410fcba2ce5349915d0df100909a294eeaa9925f and https://github.com/antiprism/mpd_oled/commit/d9644140ac99f76e1ebb87530e8cf117f5aa4e77

closed 2020-02-27 by antiprism