antiprism/mpd_oled #15: RaspDAC Mini with moOde 5.x install

rikardo1979, 2019-05-05:
Just wondering if there is a chance to make this working on the latest RaspDAC Mini running moOde?
Display used in this kit is an SPI OLED but not sure what make or model.
I may take it apart and check if needed 

Thanks in advance

antiprism, 2019-05-05:
Hi Rikardo

I recently updated mpd_oled to make it compatible with moOde 5.0.

Regarding the display, SPI isn't well supported and involves modifying the mpd_oled source to include the configuration, specifically the init_display function in display.cpp. Also, I cannot confirm that other changes won't be needed as I have had no feedback from anyone that has used mpd_oled with SPI. If the display can also work with i2c then that would be a better option.

 The controller chip also needs to be supported by mpd_oled, and should be SSD1306 or SH1106, so it might be an idea to check that first.

Adrian.

rikardo1979, 2019-05-06:
Thanks for your reply here. I do have the display working on moOde like this http://tinyw.in/rCv2

I will take it a part and find more details about it

antiprism, 2019-05-06:
Looking at the configuration file from the OLED package referenced in your link, you have
```
from luma.oled.device import ssd1306
import RPi.GPIO as GPIO
serial = spi(port=0, device=0, gpio_DC=27, gpio_RST=24)
device = ssd1306(serial, rotate=2)
```

In which case...

In main.cpp, add a line near the beginning of the main function to override whatever OLED you set with -o, which should then ensure SPI initialisation for your display
```
OledOpts opts;
opts.process_command_line(argc, argv);
opts.oled = OLED_ADAFRUIT_SPI_128x64;  // ADD THIS LINE
```
In display.cpp, add the SPI configuration
```
bool init_display(ArduiPi_OLED &display, int oled, unsigned char i2c_addr,
    int reset_gpio, bool rotate180)
{
 
// SPI
  if (display.oled_is_spi_proto(oled)) {
    // SPI change parameters to fit to your LCD
    if ( !display.init(27, 24, OLED_SPI_CS, oled) )     // INCLUDE THESE SETTINGS
      return false;
  }
```

I am not sure if OLED_SPI_CS will need changing.

I haven't built these changes, so they may include syntax errors. Also, even with the correct initialisation there may be other changes required to get this display working.

Adrian.

antiprism, 2019-05-06:
Maybe your OLED isn't connected like the original configuration in the OLED package, as is the case in this message: http://moodeaudio.org/forum/showthread.php?tid=664&pid=9235#pid9235

In which case, edit the display.init() function cal  according to your configuration (the first argument is the DC GPIO number, and the second is the RESET GPIO number).

antiprism, 2019-11-17:
Closing. Reopen if the issue remains.

closed 2019-11-17 by antiprism