antiprism/mpd_oled #16: Oled low brightness.

usstrike, 2019-05-12:
Is it possible to adjust the brightness of the screen?

antiprism, 2019-05-12:
Hi

There is no mpd_oled option to control the brightness.

From a quick search, I am unsure if it is possible to implement this properly in software. There appears to be a command for SSD1306 to control contrast, but it seems that it only has a limitted effect on brightness, and the full brightnes range involves a variable resistor (maybe a variable resistor on its own is enough).

There are some interesting threads here
   https://forum.arduino.cc/index.php?topic=515370.0
   https://forums.adafruit.com/viewtopic.php?f=47&t=38569

Adrian.

arthurlutz, 2019-12-16:
I've managed to see dimming oled using https://flows.nodered.org/node/node-red-contrib-oled (with the "Dimmed" block). Down under it is a javascript library : https://github.com/baltazorr/oled-i2c-bus/blob/master/oled.js#L340

antiprism, 2019-12-16:
Hi Arthur

The javascript code refers to "contrast", so I imagine this works the same way as the dimming function included in the mpd_oled OLED driver code, in which case setting the contrast to 0 in mpd_oled might work to dim an SSD1306 screen.

I  have ordered an SSD1306 screen so I should be able to test this soon.

Adrian.

supercrab, 2021-03-10:
I've noticed during testing that setting some of the OLED types gives differing brightness values.  I think there's a command and some data that gets sent to the screen during initialisation that could be used to support this.

i'll do some digging.

antiprism, 2021-03-10:
Hi Mase

Thanks for looking at this. I am in the processes of changing mpd_oled to use the U8g2 library, and this provides a setContrast function

   https://github.com/olikraus/u8g2/wiki/u8g2reference#setcontrast

I imagine it will do all that can be done in software to change the brightness. I will add an option for it.

Adrian.

supercrab, 2021-03-10:
U8g2 is great, the only trouble is that it takes a while to compile.  Have you considered compiling on a non Pi and dropping the binary over?

Anyway, I found it.  It's in `ArduiPi_OLED.cpp:667`  there is a method to set the brightness `void ArduiPi_OLED::setBrightness(uint8_t Brightness)`

I might implement this in my older branch, if I get time.

supercrab, 2021-03-10:
Bah, I tested this but it made no difference to the brightness :(

antiprism, 2021-03-10:
Hi Mase

The total build time for U8g2 library is less than 10 minutes on a Pi Zero, which isn't too bad, but it doesn't really affect development time as once it has been built mpd_oled just links to it.

I have sometimes cross-compiled on the Pi 4 and copied the binary over to the Pi Zero to test, but usually for bigger changes I write the code on the Pi 4, but only to ensure a successful build, and then copy over the code to the Pi Zero and change it there to make it work.

The brightness function may work for the SSD1306, but I don't think you can change the brightness of the SSH1106 using software.

Adrian.

supercrab, 2021-03-10:
OK brightness has no effect but precharge setting like you suggested in your links does work :)

supercrab, 2021-03-10:
Where there is a `SSD1306_Set_Precharge_Period` command the value after defines the brightness 0 - 34

supercrab, 2021-03-11:
@arthurlutz I have implemented contrast and pre charge period control in my fork of `mpd_oled`. 

I did this as an experiment for testing purposes but works pre charge is `-e 0 - 34` and brightness is `-t 0-255

https://github.com/supercrab/mpd_oled