antiprism/mpd_oled #17: Elapsed song time

hohmik, 2019-05-30:
Thank you for excellent plug in. Easy installation, works grate on moodeaudio.
Is it possible to replace current time on main screen with elapsed song time ?
Thank you.

antiprism, 2019-05-31:
Hi hohmik

There are currently no configuration options to change the items that are displayed. However, you could make this small change in code.

In main.cpp, around line 366, replace the lines that draw the clock

```
  int clock_offset = (disp_info.clock_format < 2) ? 0 : -2;
  draw_time(display, 128-10*W+clock_offset, 2*H, 2, disp_info.clock_format);
```
With lines to draw the elapsed time

```
   int secs = disp_info.status.get_elapsed_secs();
   string elapsed_str = msg_str("%5d:%02d", secs / 60, secs % 60);
   draw_text(display, 128-8*W-1, 2*H, 8, elapsed_str.c_str());
```

Adrian.

hohmik, 2019-05-31:
Thanks for quick response.
This small change in code work good, however elapsed song time became as small as song title.
Is there some way to make it big ?
I was trying to replace some code in display.cpp in draw_time function with some of your new code with no luck.
Even if instead of time it will be permanently displayed elapsed song time, it's OK.
Do you have any suggestions ho to do it ?
Thank you.
[image]

antiprism, 2019-05-31:
The characters are written with a bitmap font, and the the text size for the elapsed time will either be the small size or the larger clock text size, it cannot be a size in between. Unfortunately, the larger text will only allow two digits for the minutes of elapsed time in the space available, which is 99 minutes maximum, and this will easily be exceeded if listening to internet radio.

Adrian.

hohmik, 2019-05-31:
It acceptable if 99 minutes will be exceeded. For for those who listen only music files it is not a problem.
If you can, just tell me what replace in display.cpp in draw_time function.
Thanks a lot.

antiprism, 2019-05-31:
The repacement code for the larger text can be

```
int secs = disp_info.status.get_elapsed_secs();
string elapsed_str = msg_str("%2d:%02d", secs / 60, secs % 60);
elapsed_str.resize(5);
display.setCursor(128-10*W, 2*H);
display.setTextSize(2);
for(int i=0; i<5; i++)
   display.write((uint8_t)elapsed_str[i]);
```

Adrian.

hohmik, 2019-05-31:
Superb! Work like a charm.
Here is my audio player
[https://youtu.be/Oetar3SE1Mw](https://youtu.be/Oetar3SE1Mw)

antiprism, 2019-05-31:
Looks good! Nice job.

Adrian.,

closed 2019-06-02 by hohmik