antiprism/mpd_oled #19: Not compiling on buster

wildestpixel, 2019-07-01:
Able to compile on Stretch but not Buster. Output at make stage on buster :

pi@quattro:~/mpd_oled $ PLAYER=MPD make
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o main.o main.cpp
In file included from /usr/include/c++/8/vector:69,
                 from display_info.h:29,
                 from display.h:32,
                 from main.cpp:25:
/usr/include/c++/8/bits/vector.tcc: In member function 'void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const double&}; _Tp = double; _Alloc = std::allocator<double>]':
/usr/include/c++/8/bits/vector.tcc:413:7: note: parameter passing for argument of type 'std::vector<double>::iterator' {aka '__gnu_cxx::__normal_iterator<double*, std::vector<double> >'} changed in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/vector:64,
                 from display_info.h:29,
                 from display.h:32,
                 from main.cpp:25:
/usr/include/c++/8/bits/stl_vector.h: In constructor 'OledOpts::OledOpts()':
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<double*, std::vector<double> >' changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<double*, std::vector<double> >' changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<double*, std::vector<double> >' changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<double*, std::vector<double> >' changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h: In member function 'virtual void OledOpts::process_command_line(int, char**)':
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<double*, std::vector<double> >' changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<double*, std::vector<double> >' changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<double*, std::vector<double> >' changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type '__gnu_cxx::__normal_iterator<double*, std::vector<double> >' changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o timer.o timer.cpp
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o status.o status.cpp
status.cpp: In member function 'std::__cxx11::string mpd_info::get_elapsed_time() const':
status.cpp:410:36: warning: '%02d' directive output may be truncated writing between 2 and 3 bytes into a region of size between 1 and 6 [-Wformat-truncation=]
   snprintf(time_str, time_str_len, "%02d:%02d:%02d", hours, mins, secs);
                                    ^~~~~~~~~~~~~~~~
status.cpp:410:36: note: directive argument in the range [-59, 59]
status.cpp:410:36: note: directive argument in the range [-59, 59]
status.cpp:410:11: note: 'snprintf' output between 9 and 16 bytes into a destination of size 9
   snprintf(time_str, time_str_len, "%02d:%02d:%02d", hours, mins, secs);
   ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
status.cpp: In member function 'std::__cxx11::string mpd_info::get_total_time() const':
status.cpp:410:36: warning: '%02d' directive output may be truncated writing between 2 and 3 bytes into a region of size between 1 and 6 [-Wformat-truncation=]
   snprintf(time_str, time_str_len, "%02d:%02d:%02d", hours, mins, secs);
                                    ^~~~~~~~~~~~~~~~
status.cpp:410:36: note: directive argument in the range [-59, 59]
status.cpp:410:36: note: directive argument in the range [-59, 59]
status.cpp:410:11: note: 'snprintf' output between 9 and 16 bytes into a destination of size 9
   snprintf(time_str, time_str_len, "%02d:%02d:%02d", hours, mins, secs);
   ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
status.cpp: In member function 'std::__cxx11::string mpd_info::get_kbitrate_str() const':
status.cpp:434:26: warning: '%4d' directive output may be truncated writing between 4 and 11 bytes into a region of size 5 [-Wformat-truncation=]
   snprintf(str, str_len, "%4d", rate);
                          ^~~~~
status.cpp:434:26: note: directive argument in the range [-2147483648, 9999]
status.cpp:434:11: note: 'snprintf' output between 5 and 12 bytes into a destination of size 5
   snprintf(str, str_len, "%4d", rate);
   ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 18 lines cut ...]
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o display.o display.cpp
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o programopts.o programopts.cpp
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o ultragetopt.o ultragetopt.cpp
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o ArduiPi_OLED.o ArduiPi_OLED.cpp
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o Adafruit_GFX.o Adafruit_GFX.cpp
cc  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o bcm2835.o bcm2835.c
bcm2835.c: In function 'bcm2835_i2c_write':
bcm2835.c:877:12: warning: implicit declaration of function 'i2c_smbus_write_byte_data' [-Wimplicit-function-declaratio]
   reason = i2c_smbus_write_byte_data(i2c_fd, buf[0], buf[1]);
            ^~~~~~~~~~~~~~~~~~~~~~~~~
bcm2835.c:880:12: warning: implicit declaration of function 'i2c_smbus_write_word_data' [-Wimplicit-function-declaratio]
   reason = i2c_smbus_write_word_data(i2c_fd, buf[0], (buf[2]<<8) | buf[1] );
            ^~~~~~~~~~~~~~~~~~~~~~~~~
bcm2835.c:883:12: warning: implicit declaration of function 'i2c_smbus_write_i2c_block_data' [-Wimplicit-function-declaration]
   reason = i2c_smbus_write_i2c_block_data(i2c_fd, buf[0], len-1, (const __u8 *) &buf[1]);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
g++  -o mpd_oled main.o timer.o status.o status_msg.o utils.o display.o programopts.o ultragetopt.o ArduiPi_OLED.o Adafruit_GFX.o bcm2835.o -lmpdclient -lpthread 
/usr/bin/ld: bcm2835.o: in function `bcm2835_i2c_write':
bcm2835.c:(.text+0x27d0): undefined reference to `i2c_smbus_write_i2c_block_data'
/usr/bin/ld: bcm2835.c:(.text+0x27ec): undefined reference to `i2c_smbus_write_word_data'
/usr/bin/ld: bcm2835.c:(.text+0x27fc): undefined reference to `i2c_smbus_write_byte_data'
collect2: error: ld returned 1 exit status
make: *** [Makefile:28: mpd_oled] Error 1

antiprism, 2019-07-01:
Hi

Thanks for the build report.

There are 3 kinds of notification in the output.

1. The lines with "changed in GCC 7.1" are a note indicating , I believe, that if you link with libraries built with an earlier version of GCC then there is a potential for errors.

2. The lines with "directive output may be truncated" are warnings, due to the mpd_oled code assuming that certain values will never be negative. I'll try to enforce the positive values to remove the warnings.

3. The build is failing because of

```
g++  -o mpd_oled main.o timer.o status.o status_msg.o utils.o display.o programopts.o ultragetopt.o ArduiPi_OLED.o Adafruit_GFX.o bcm2835.o -lmpdclient -lpthread 
/usr/bin/ld: bcm2835.o: in function `bcm2835_i2c_write':
bcm2835.c:(.text+0x27d0): undefined reference to `i2c_smbus_write_i2c_block_data'
```

It looks like on your system the i2c functions are in a library that needs to be specified when linking. You can hopefully link in the library by building mpd_oled with

`LDFLAGS="-li2c" make`

Adrian.

wildestpixel, 2019-07-01:
Hi,

Many thanks for getting back ref the situation with buster, that’s a fantastic level of support for your fantastic project which is appreciated by all.

What would the command line be as a whole to build for MPD and specify the i2c library ? Or would the line you have given be sufficient ?

antiprism, 2019-07-01:
Hi

I have just installed GCC 8 on Ubuntu and I can see that the line I posted includes the library in the wrong place, and so does not fix the build error. The variable should really be LDLIBS, but the mpd_oled Makefile needs to be fixed to support this.

The i2c library nees to be included when building on RuneAudio, and there are no other changes with respect to a vanilla MPD build, so for now you should be able to build everything with

`PLAYER=RUNEAUDIO make`

Adrian

antiprism, 2019-07-01:
Hi

You may need to install libi2c-dev (and also i2c-tools and lm-sensors) , as I have just noticed that you have lines in your output like "warning: implicit declaration of function 'i2c_smbus_write_byte_data'", meaning also that suitable i2c library headers are not being included. With these packages installed you may or may not need to link with -li2c, so try `PLAYER-RUNEAUDIO make` and `PLAYER=MPD`. Also, run `make clean` between builds.

If this doen't work out, I'll see if I can find an SD card and put a copy of buster on it to look at what is needed, as this will certainly become an issue for more people.

Adrian.

wildestpixel, 2019-07-01:
I can confirm that `PLAYER=RUNEAUDIO make` works well and now has a result the same for me as when running with the MPD flag

antiprism, 2019-07-02:
Great, I am pleased it is working!

I managed to find an SD card and worked through the mpd_oled install on raspbian buster. I can see that there are still warnings for the i2c_smbus_* functions, because they do not appear in any included header, but everything else is in place and the build completes correctly.

I will update the docuimentation for the debian build and close the issue.

Thanks again for letting me know about this.

Adrian.

closed 2019-07-03 by antiprism