antiprism/mpd_oled #20: error compiling on volumio (community build) orangepi-pc

VincentGijsen, 2019-07-03:
```
volumio@volumio:~/mpd_oled$ PLAYER=VOLUMIO make
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DVOLUMIO  -c -o status.o status.cpp
status.cpp: In function 'std::string secs_to_time(int)':
status.cpp:400:10: error: 'abs' is not a member of 'std'
   secs = std::abs(secs);
          ^
status.cpp:400:10: note: suggested alternative:
In file included from status.cpp:41:0:
/usr/include/stdlib.h:775:12: note:   'abs'
 extern int abs (int __x) __THROW __attribute__ ((__const__)) __wur;
            ^
status.cpp: In member function 'std::string mpd_info::get_kbitrate_str() const':
status.cpp:432:23: error: 'abs' is not a member of 'std'
   int rate = std::min(std::abs(kbitrate), 9999);
                       ^
status.cpp:432:23: note: suggested alternative:
In file included from status.cpp:41:0:
/usr/include/stdlib.h:775:12: note:   'abs'
 extern int abs (int __x) __THROW __attribute__ ((__const__)) __wur;
            ^
<builtin>: recipe for target 'status.o' failed
make: *** [status.o] Error 1
volumio@volumio:~/mpd_oled$
```

```
volumio@volumio:~/mpd_oled$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Debian 4.9.2-10+deb8u1' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.9.2 (Debian 4.9.2-10+deb8u1)

````

antiprism, 2019-07-04:
Hi

Thank you for reporting the issue.

I have just updated the repository with changes that hopefully fix this error.

Adrian.

antiprism, 2019-07-28:
Closing, due to no new information.

closed 2019-07-28 by antiprism