antiprism/mpd_oled #21: Posible to use multiple displays

nocturneop15, 2019-07-24:
could this great piece of software be modified to display only selected information?

My idea is to support more displays. I would like to use 
- left dispay for left channel spectrum analyzer 
- center display for song, artist, clock  and other text info 
- right display for right channel spectrum analyzer 

Idea of using cheap displays is great, but those displays are so cheap, that using more of them can provide much more comfort and fanciness :-) 

Sorry for placing this to "isssues", I did not found better place for feature suggestion.

supercrab, 2019-07-24:
I had a similar idea and bought some 256 x 64 pixel OLED displays so I could play with doubling the width of the screen.  I'm currently coding the screen driver for this right now.  I just wanted to add album art (in monochrome) and spread things out a bit, plus it gives me something to do in the evenings.

I spoke to antiprism and he does intend to support custom layouts at some point.  This might be a JSON file(s) that would allow the layout to be user configurable.   You could have everything located and sized just the way you want it to be.  Really nice idea especially for displays of different sizes.

Anyway, back to multiple displays.  There's 2 types of OLED SPI and I2C.  Normally, I2C OLEDs have a 0ohm resistor on them that allow you to change the address of the device between one value or another.  This would allow you to easily hook up 2 OLEDs to your device (we're talking about the cheap displays here).  
You should be able to run 2 instances of mpd_oled twice and by specifying a different I2C address for each display.  If this is coupled with the customisable layout functionality then you would have 2 different displays quite easily.

You could have up to 32 SPI screens with this custom PI SPI driver.  I don't know much about it but I did see 5 screens going at once!  See here: http://www.newsdownload.co.uk/pages/RPiSPiDev.html

Currently, it's not possible to have more than 1 diplsay over SPI.  If mpd_oled were to support this it would take some work I reckon!

antiprism, 2019-07-25:
Hi nocturneop15

Thanks for the feedback and suggestions (it is fine to use the issues section for this).

I have already started work on custom display layouts, but I am unlikely to be able to continue to looking at this before the end of September. My current plans are https://github.com/antiprism/mpd_oled/issues/18#issuecomment-499982540

Someone is already running two instances of mpd_oled, with some slight code changes, on I2C displays. One display shows the active player screen and the other shows the idle screen.

@supercrab, the technique to have a lot of SPI displays looks interesting. I wonder how many screens can share an SPI or I2C bus and still allow a frame rate high enough for the spectrum display? From a quick search it looks like it might be possible to have two (or three) SPI devices without any extra hardware https://www.raspberrypi.org/forums/viewtopic.php?t=140159, and it looks like u8g2 supports this https://github.com/olikraus/u8g2/issues/388 .

Adrian.

supercrab, 2019-07-25:
Having 1 screen is fine for me! :)

nocturneop15, 2019-07-26:
antiprism: great you think about implementing custom layouts, everyone will find his best :-) In the meantime, I will order some more displays and try some tricks you supercrab mentioned.  

As I look to code, it looks very readable even when I have no programing skills (except powershell). Maybe it is a good time to start learning more complex programming :-)

supercrab, 2019-07-27:
Good luck! :)