antiprism/mpd_oled #22: display line overflow

nocturneop15, 2019-07-24:
Hi again, 
I just notticed that in iddle state, network icon next to IP address is overflowing - far right 1px column is rendered far left, row just below the first. 

 Everything else works normally, so this is just a small cosmetic glitch.
[image]

supercrab, 2019-07-24:
I had this when I was using the wrong display command line parameter o 6. Try o 3

antiprism, 2019-07-25:
Hi nocturneop15

I just checked my display with the ethernet icon and the display didn't wrap for me. Hopefully supercrab's suggestion will fix this for you.

Adrian.

nocturneop15, 2019-07-26:
Hi,
supercrab is true! with 
`./mpd_oled -o 6 -b 10 -a 3c -g 1 -f 30`, display has 1 line overflow and 90% of time does not start automatically after unclean shutdown. I had to start program manually, screen started to flicker, then I close the program. Flickering stopped and display remained under service control working good, buth with 1 line overflow. 

Using `./mpd_oled -o 6 -b 10 -a 3c -g 1 -f 30`, everything is perfect. 

Thank you both for support!

nocturneop15, 2019-07-26:
Mistype. By using`./mpd_oled -o 3 -b 10 -a 3c -g 1 -f 30`, everything is perfect :-)

closed 2019-07-26 by nocturneop15