antiprism/mpd_oled #23: screensaver option

nocturneop15, 2019-07-27:
Hello again, 
I tried to implement screensaver by entering -S parameter to mpd_oled. I had a partial success, but now, I'm stuck and kindly asking for help

Basic idea was simple - if player is stopped, paused (or unknown) and -S parameter provided, mpd_oled should draw "draw_screensaver" function instead of "draw_clock". 

-S parmeter should toggle "screensaver" variable to "true", which initialy set to "false"

My problem is - it does not switch it. When I compile it with "true", it works as expected, but I cannot get switching by parameter working. I think this will only need some small tweak, but this is my first try to programming and I still do not understand all syntaxes fully. 

main.cpp: 
[main.txt](https://github.com/antiprism/mpd_oled/files/3437920/main.txt)

Maybe you can find this usefull too. It willl save OLED display life, and also provide darkness for sleeping if you have your player in a bedroom.

antiprism, 2019-07-27:
Hi

You could (untested) define the draw display function like this

```void draw_display(ArduiPi_OLED &display, const display_info &disp_info, bool screensaver)```

and use the `screensaver` variable in the function to test if the screensaver is being used. Call it, in start_idle_loop(), like this

```draw_display(display, disp_info, opts.screensaver);```

I have easily visible screen burn on my display, but I like having the clock.

Adrian.

nocturneop15, 2019-07-27:
Thanks, I was so close :-D 

Now, it works as expected, when -S is defined, display is black when nothing is playing. 
If -S parameter is not there, it behaves as usual, clock is displayed together with connection details 

[main.txt](https://github.com/antiprism/mpd_oled/files/3438466/main.txt)

I'm not sure how such unauthorized code changes should be handled, I hope it is OK to post this here with my changes. 

For the future, I think about having multiple screensaver options:
- black screen for empty playlist or paused song (corrently works)
- scrolling song name for paused song
- screensaver on scheduled basis (song info displayed only some time after new song started, than going back to black 

I live in a deep forrest, so I like enjoying absolute darkness, absolute silence - and if - absolute quality sound of my music :-)

antiprism, 2019-07-28:
Hi

It is fine to post specific code changes here. For ongoing or more general changes, it is better to share them through a fork of the project.

Sounds like you have an ideal listening environment!

Adrian.

antiprism, 2022-11-05:
Closing. Collecting screensaver issues in https://github.com/antiprism/mpd_oled/issues/72

closed 2022-11-05 by antiprism