antiprism/mpd_oled #24: Moode- not compiling

superjasiek, 2019-08-10:
Hi thanks for this nice piece of software. Unfortunately there is similar error like in runeaudio

g++  -o mpd_oled main.o timer.o status.o status_msg.o utils.o display.o programopts.o ultragetopt.o ArduiPi_OLED.o Adafruit_GFX.o bcm2835.o  -lmpdclient -lpthread
/usr/bin/ld: bcm2835.o: in function `bcm2835_i2c_write':
bcm2835.c:(.text+0x27d0): undefined reference to `i2c_smbus_write_i2c_block_data'
/usr/bin/ld: bcm2835.c:(.text+0x27ec): undefined reference to `i2c_smbus_write_word_data'
/usr/bin/ld: bcm2835.c:(.text+0x27fc): undefined reference to `i2c_smbus_write_byte_data'
collect2: error: ld returned 1 exit status
make: *** [Makefile:28: mpd_oled] Error 1

is there any chance to compile mpd_oled correctly?

Take care!

antiprism, 2019-08-10:
Hi

Thanks for reporting the issue. Which version of Moode are you building on?

For now, you may be able to build mpd_oled with

  `PLAYER=MOODE LDFLAGS="-li2c" make`

Adrian.

superjasiek, 2019-08-10:
Hi!
Thank you for quick reply :-) I use Moode 6.0  on raspbian buster. Your solution worked like charm :-)
Thank you very much
Jan

antiprism, 2019-08-11:
Great, thanks for letting me know! I'll look at updating the install for Moode 6.

Adrian.

closed 2019-08-15 by antiprism