antiprism/mpd_oled #25: Spectrum delay / error in mpd_oled.service

jfaragher, 2019-08-13:
Successfully installed for Volumio, with the following issues

1.  I'm unable to sync the spectrum with the sound.  I've changed the #buffer_time in /volumio/app/plugins/music_service/mpd/mpd.conf.tmpl significantly higher and lower, and resaved audio, and it seems to make no difference?

2.  Seems to be a missing linebreak in mpd_oled.service after [Service] which prevented the service from working, gave "Failed to execute operation: Bad message" error.  Fixed with manual editing!

antiprism, 2019-08-14:
Hi

Thank you for reporting the issues

1. I believe that the 'buffer_time' setting does nothing, and I will be removing references to it from the documentation. You could try changing the the audio buffer size in Volumio (I am unsure if this helps). A delay can be introduced by some audio hardware, amd there is currently no mpd_oled option to correct this. Please see: https://github.com/antiprism/mpd_oled/issues/18#issuecomment-500143890 (and replies)

2. I haven't been able to find a problem with the mpd_oled.service file. I cloned a fresh copy of the repository. Looking at a hex dump of the file the line termination seems fine. The file displays with a newline at github https://raw.githubusercontent.com/antiprism/mpd_oled/master/mpd_oled.service .

Adrian.

antiprism, 2019-08-14:
> 1. I believe that the 'buffer_time' setting does nothing, and I will be removing references to it from the documentation.

I have updated the repository with this change.

antiprism, 2019-11-17:
Closing. Reopen if the issue remains.

closed 2019-11-17 by antiprism