antiprism/mpd_oled #26: Spectrum bars doesn't work

hueh, 2019-09-23:
I want to thank you for you great work with oled screen ;)
I have one problem and I don't know how to fix this. Spectrum bars doesn't work:/ I attached photo how it looks now. Can you help me?

[image]

antiprism, 2019-09-23:
Hi

It is difficult, but possible, to make the spectrum display work with Spotify on Volumio. See https://github.com/antiprism/mpd_oled/issues/4#issuecomment-423255641 (and the reast of that issue).

In your image, on the left of the spectrum area is small vertical bar. This might be the wifi signal display having wrapped, which can happen if you have not specified your display type correctly with
'mpd_oled -o'. See https://github.com/antiprism/mpd_oled/issues/22

Adrian.

antiprism, 2019-11-17:
Closing. Re-open if the issue remains.

closed 2019-11-17 by antiprism