antiprism/mpd_oled #27: SSD1322 Support

Maschine2501, 2019-11-16:
Hi! 
At first, great work!!!
I want to use mpd_oled with a 256x64 SPI SSD1322 Oled.
I got the driver File with all adresses ready, but im not sure how to implement it.
In your tutorial for Volumio you have made this quote: "(For, SPI OLEDs, edit display.cc to include your connection details, if this works out I will provide options for these parameters.)"
When installed, there are 4 Files named "display", ending on: .cpp .h .o and display-info.h .
I think that i'm able to fill the adresses and parameters, but i don't know the dependencies, or in other words: i don't know in which files i've to change what....
I hope you could help me to solve this problem,
best ragards!

antiprism, 2019-11-16:
Hi

The file that needs to be changed to enable SPI is display.cpp. This file was renamed from display.cc, so thanks for reporting this. I will update the documentation with the new file name.

However, your display is not supported by mpd_oled so the changes would involve more than just entering the SPI configuration, and even if you are successsful then mpd_oled will only display on half the screen, unless you make further changes to the code.

@supercrab might be able to comment on this, as I believe he has looked at modifying the mpd_oled code to support your display.

Adrian.

supercrab, 2019-11-16:
Hi

Yes, I did manage to get this working with the same screen as you but the refresh rate was about 9 FPS on my Pi 0. It was too slow for it to be of any use because I wanted a nice and snappy equaliser.  

I knocked the idea of using the head and decided to use a monochrome 2 inch monochrome I2C screen and that worked like a dream.

I can upload the code if you’re interested though

Cheers
Mase

Maschine2501, 2019-11-17:
@antiprism Thank you for your feedback!
I asumed that i also had to change other parts of the code... but isn't the way the destination? 😅
How hard this would be? But who doesn't try, won't succseed.

@supercrab Thank you for your feedback, i indeed would be happy to get the modified code!
I'm using a raspberry pi 4 /4gb for my project, that should  bring more than 9fps...
So i hope i can test your modified code.

Great Comunity spirit, thank you guys!

Maschine2501, 2019-11-19:
@supercrab Hi, do you need my e-mail adress to send me the promised code?
I'm super curious to try it!
Greetings!

supercrab, 2019-11-19:
Yes please I’ll zip it up for ya!

Maschine2501, 2019-11-19:
@supercrab Good Evening! The mail adress is "o-z @ gmx.de" without the spaces before and after the @.
 Thank you!

Maschine2501, 2019-11-20:
@antiprism Thank you for bringing me in touch with @supercrab he will send me his modified Code.
So the Incident is solved!
Thanks, guys!

closed 2019-11-20 by Maschine2501

supercrab, 2019-11-20:
Extra notes for SSD1322 support.  You'll need to enable 4 wire SPI on the screen.  This will require moving a 0ohm resistor or 2.  This is because the DC pin is used to determine if data or command is being transmitted.