antiprism/mpd_oled #32: Support for changing disposition to avoid burn of pixels

arthurlutz, 2019-12-16:
It would be nice to rotate accros 2 or 3 ways of displaying the information so as to avoid burning pixels. If the same time zone or track zone is displayed constantly the screen will "damage" those zones.

antiprism, 2019-12-16:
Thanks for reporting this. I have also seen it on my display, and I think it is worth addressing. One approach is to invert the display periodically (i.e. black on white rather than white on black), which I think would be straigtforward to add in.

Adrian.

antiprism, 2019-12-16:
I have added a new option -I that inverts black and white, with or without a periodic change: https://github.com/antiprism/mpd_oled/commit/deae92feea21ca2668581487add8f00cf7706c83

I don't plan to add a periodic display of alternative information screens to mpd_oled, and so I am closing the issue, but I will consider whether this can be supported in future projects.

Adrian.

closed 2019-12-16 by antiprism