antiprism/mpd_oled #33: RPi 3B Volumio compilation problems

misko903, 2019-12-28:
Hello antiprism,
I have problem with installing Cava on my Volumio. 
Can you help me, please?

some months ago, it was working well, I have updated Volumio to latest 2.692 ant i have ended on the command below:

```
volumio@volumio:~/cava$ sudo make install
[sudo] password for volumio:
Making install in iniparser
make[1]: Entering directory '/home/volumio/cava/iniparser'
make[2]: Entering directory '/home/volumio/cava/iniparser'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/volumio/cava/iniparser'
make[1]: Leaving directory '/home/volumio/cava/iniparser'
make[1]: Entering directory '/home/volumio/cava'
/bin/bash ./libtool  --tag=CC   --mode=link gcc -std=gnu99 -std=c99 -Wall -Wextr                               a -Wno-unused-result -Wno-maybe-uninitialized  -g -O2 -L/usr/local/lib -Wl,-rpat                               h /usr/local/lib  -o cava cava-cava.o cava-config.o input/cava-common.o input/ca                               va-fifo.o output/cava-terminal_noncurses.o output/cava-raw.o input/cava-alsa.o                                     -liniparser -Liniparser/.libs -lpthread -lasound -lm -lfftw3
libtool: link: gcc -std=gnu99 -std=c99 -Wall -Wextra -Wno-unused-result -Wno-may                               be-uninitialized -g -O2 -Wl,-rpath /usr/local/lib -o cava cava-cava.o cava-confi                               g.o input/cava-common.o input/cava-fifo.o output/cava-terminal_noncurses.o outpu                               t/cava-raw.o input/cava-alsa.o  -L/usr/local/lib /home/volumio/cava/iniparser/.l                               ibs/libiniparser.a -Liniparser/.libs -lpthread -lasound -lm -lfftw3
input/cava-alsa.o: In function `initialize_audio_parameters':
/home/volumio/cava/input/alsa.c:24: undefined reference to `alloca'
collect2: error: ld returned 1 exit status
Makefile:534: recipe for target 'cava' failed
make[1]: *** [cava] Error 1
make[1]: Leaving directory '/home/volumio/cava'
Makefile:787: recipe for target 'install-recursive' failed
make: *** [install-recursive] Error 1
```

antiprism, 2019-12-29:
Hi Michal

Thank you for reporting this issue. I have replied to the corresponding issue at the Cava project: https://github.com/karlstav/cava/issues/331

Adrian.

antiprism, 2019-12-31:
Fixed in Cava

closed 2019-12-31 by antiprism

misko903, 2020-01-01:
Thanks and happy new year from Slovakia!