antiprism/mpd_oled #34: OLED SSD1309 Display will not work on Volumio

Korniman-the-Rooky, 2020-02-21:
Dear Adrian, I use an Raspi 4 with Volumio. All is working on the latest software. Now I found your really cool project. Unfortunately I am not so deep into Linux, but copy and paste is working well :-). I bought the following Display: 1,54 ZOLL OLED128X64 SSD1309 which should be compatible with SSD1306
[https://www.ebay.de/itm/253369133648](https://www.ebay.de/itm/253369133648) 
The display will not work, after several days of work. I think I need your help. 
I read all forum content in the different platforms, but without success.
I cannot use I2S because the port is already booked and running with a RTC. I checked the 1.54 SSD1309 OLED Display function with an Arduino project from my coffee machine and the display is working there when I exchanged it from its old 0.96 SSD1306 OLED from the first moment.
Ok, back to the Volumio DAC:
I activated the SPI function in the config.
I configured the display as following on the Raspi GPIO (Pin No is mentioned not GPIO No.)

OLED Pin - Raspi Pin
1: GND - 14: GND
2: VCC - 9: 3,3V
3: SCL - 23: SPI SCLK
4: SDA - 19: SPI MOSI
5: RES - 22: GPIO 25
6: DC - 18: GPIO 24
7: CS - 24: SPI CS0

I set your code to Adafruit SPI 128x64.
"sudo ./mpd_oled -o 3 -b 10 -g 1 -f 20
The Installation guide from you did work fine until the following lines:

"If you have a reset pin connected, specify the GPIO number with mpd_oled -r, e.g. mpd_oled -o6 -r 24 .... (For, SPI OLEDs, edit display.cpp to include your connection details, if this works out I will provide options for these parameters.)"

Please can you support me? 
I have no clue how to configure the last needed settings.
Your manual is wonderfull for a rooky like me until the last lines. Please, can you write on the end of your manual clear commands?
Is my Pin setting on the GPIO wrong?
Where and How can I configure these settings in your code?
Did you try out the SPI display or did anyone tried this out with success?
What further data do you need for support?

Currently my display is completely blank and there is not poping up only one pixel :-)

Thank you in advance and best regards
Benjamin from Frankfurt

antiprism, 2020-02-22:
Hi Benjamin

Thank you for your message. There are no command line SPI options (I only had an I2C display until recently and so could no test them). The code itself supports SPI, but using it requires changing the code. See the following issue (success not confirmed): https://github.com/antiprism/mpd_oled/issues/15 (you would need to specify your particular pins and also use OLED_SPI_CS0). The following issue confirms a working SPI display (but includes extra changes for the SSH1106 controller): https://github.com/antiprism/mpd_oled/issues/14

As I now have an SSD1306 SPI display I can look at providing an option for it. I'll take a look at what is involved and then post an update (probably within a couple of days).

Adrian.

Korniman-the-Rooky, 2020-02-22:
Dear Adrian, thank you for your fast feedback. Excellent.
Regarding Link #15 : My Pin setting is exactly the same as in the linked moode forum. I also believe that it is a configuration issue in the display.init., with CS, Reset and DC, maybe SPI is not in display mode, but I am not sure what should be written in correctly. Further I have no Idea what will be the correct frame rate.
Regarding Link #14. I also have an additional 1.3 SPI SSD1106 OLED display, if 1.54 SPI SSD1309 will finally not work, but 1.54 is much larger which looks very cool. The guy who tried out SSD1106 has fixed so much in the code that for me it is not clear if all of these changes are necessary and it is not clear documented for a rooky like me. Thank you for your idea to try it out with SSD1306 which should work like a SSD1309. This will hopefully bring exact repeatable results. I will wait for your post. Thank you so much. Cheers Benjamin

antiprism, 2020-02-24:
Hi Benjamin

I have got my SPI display up and running!

Initially it looked like image 2A in the following thread, but with some text writng and flashing on the top line: https://forum.arduino.cc/index.php?topic=265557.0

After reading the thread, I checked what I had bought on eBay. There was a 0.96" and 1.3" SPI display in the listing, which I thought differed only in size, but the smaller one had a SSD1306 controller and the larger one, which I bought, had a SH1106 controller!

I set up a new mpd_oled -o setting for the SPI SH1106 128x64 display and added in the code mentioned in #14, and the display is working.

I need to test the changes before updating the repository.

However, if you have an SSD1306/9 controller, you should be able to get it working now if you edit the init_display function in display.cpp and make the following change
```
// SPI change parameters to fit to your LCD
if ( !display.init(OLED_SPI_DC, reset_gpi, OLED_SPI_CS0, oled) )
  return false;
```
And run mpd_oled with -o 1 (Adafruit SPI 128x64), e.g.
```sudo ./mpd_oled -o 1 -b 21 -g 1 -f 15```

I'll add options to mpd_oled to specify DC and CS. 

Adrian.

Korniman-the-Rooky, 2020-02-24:
Dear Adrian Thank you.
To make it clear for me :-).....
1. The pin setting as mentioned above from me is correct and I can leave it as it is.
2. I have to open display.cpp and have to paste "if ( !display.init(OLED_SPI_DC, reset_gpio, OLED_SPI_CS0, oled) )
  return false;" **on the end of the code and save the file**.
3. I have to edit display settings to "sudo ./mpd_oled -o 1 -b 21 -g 1 -f 15". 
3.1. Question To your manual:
"....with a display of 10 bars and a gap of 1 pixel between bars and a framerate of 20Hz...."
What will be the minimum and maximum for variable -b, -g,  and -f ?
4. Play music and OLED display should work or must I first execute point 5.?
5. Concerning your manual text: 
"Once the display is working, edit the file mpd_oled.service to include your OLED type number with the mpd_oled command, and any other options" I have the question how to include my OLED type number with the mpd_oled comand and what kind of other 
options I have to adjust? sorry for the stupid question.
6. Execute "sudo bash install.sh"
7. Execute "sudo systemctl start mpd_oled"

May You could approve point 1-4 and 6-7 and clarify point 5. In that case I could test this tomorrow with the Raspi DAC and my SSD1309 OLED
Kind regards Benjamin

I will try it tomorrow when I have access to my Raspi DAC. Thank you so far. I let you now the results. Good evening Benjamin

antiprism, 2020-02-25:
Hi Benjamin

1. Your pins are fine. I used the same ones in testing.

2. In display.cpp there is a function called init_display that contains three lines like I posted, but the middle line ( 'if ( !display.init(...') is different. Replace this middle line with the middle line I posted. However, I wil probably update the mpd_oled repository today or tommorow so it might be easier to wait for that.

3. 4. and 5. The idea is that you run the mpd_oled command on the command line first and see how the display looks (play some music while mpd_oled is running to see the spectrum analyser working), then when you are happy with a set of options you edit the 'mpd_oled.service' file so the command there has the same options. The factors affecting -b, -g and -f are that when -b and -f are higher it requires a connection between the Pi and OLED that can run fast enough to support it, and also a fast enough Pi that handle the extra audio processing. If the settings are too high the spectum analyser won't correspond to what is playing. Higher values of -b and -g are limited by the width of the display area.

6. and 7. Yes

I'll post when I have updated the repository.

Adrian.

Korniman-the-Rooky, 2020-02-25:
Dear Adrian, thank you for your patience with me
I tried the following

0. Installation was fine, I did not changed that and did not override the OS or reinstall your code

1. Only to be 100% sure I checked the Pin Setting 3 times. It is like it is only difference is that used VCC 3,3V is Pin 17 not Pin 9

2. I exchanged the pin header wires to new ones between Raspi and OLED. Installed are 20cm cables and checked that Voltage is available on the display (3,29 Volt should be OK I asume)

3. In "raspi-config" I double checked that SPI interface is enabled

4. with cd I switched to "mpd_oled" and edited with nano the "display.cpp" here is the copy and paste from the text there (If I am correct you only changed the wording "..._CS" to "..._CS0"):

`bool init_display(ArduiPi_OLED &display, int oled, unsigned char i2c_addr,
    int reset_gpio, bool rotate180)
{

// SPI
  if (display.oled_is_spi_proto(oled)) {
    // SPI change parameters to fit to your LCD
//    if ( !display.init(OLED_SPI_DC, reset_gpio, OLED_SPI_CS, oled) )
      **if ( !display.init(OLED_SPI_DC, reset_gpio, OLED_SPI_CS0, oled) )**
     return false;
  }
  else {
    // I2C change parameters to fit to your LCD
    if ( !display.init(reset_gpio, oled, i2c_addr) )
      return false;
  }

  display.begin();
`
As you can see I only disabled your original function but I paste over your new code to be sure.

5. To be sure I switched to GUI > General Playback Options > Audio Output and selected the sound card again and saved. (My USB Toneboard is the Khadas Toneboard which works well). After that I did a reboot on the Raspi

6. I switched to directory "mpd_oled" and executed "sudo ./mpd_oled -o 1 -b 21 -g 1 -f 15" (I also tried out with "sudo ./mpd_oled -o 1 -b 10 -g 1 -f 5") and played a mp3 which started to play and I did hear it.

7. Nothing happend. 1.54 SSD1309 SPI OLED is black. After that I tried the same with my 0.96 SSD1306 SPI OLED. The same result. OLED is black

What could I do?
I also could use my 1.3 SSD 1106 SPI OLED for a test. If beneficial you could give me a hint I this should work without more changes.
Further is there a chance to see if software installation works properly? I mean all was working during the installation of your code, but I do not want to oversee something
What would be the first moment when I should see that display read anything

To give you all information 
I have to add that I run the RemotePi Board. 
[https://www.msldigital.com/collections/all-products/products/remotepi-board-for-pi-4-external-ir-and-led](url) You can activate the PCB within volumio as plugin. This is the only plugin what is running. GPIO which are used from the IR Remote PCB are:
Pin 2 and Pin 4	+ 5V
Pin 6 and Pin 9	GND
Pin 8	GPIO 14 for the shutdown signal and TxD for the serial communication
Pin 10	GPIO 15 for the shutdown signal and RxD for the serial communication
Pin 12	GPIO 18 for the IR receiver. This can be changed to GPIO 17, see below

Thank you in advance
Benjamin

antiprism, 2020-02-25:
Hi Benjamin

I believe your setup is fine.

I didn't specifically say to run 'PLAYER=VOLUMIO make' after changing the source code, but it needs to be run, and I wonder if that was the issue. If you didn't run it then sorry for not mentioning it.

I have now updated the mpd_repository with changes that should support both your SPI OLEDs. You should be able to start from:
https://github.com/antiprism/mpd_oled/blob/master/INSTALL_VOLUMIO.md#build-and-install-mpd_oled
Clone the source repository...

After you have built the program, if your DC and CS pins are as before they are default values and so don't need to be specified, in which case test mpd_oled with

SPI SSD1309: sudo ./mpd_oled -o 1 -b 21 -g 1 -f 20
SPI SSH1106: sudo ./mpd_oled -o 7 -b 21 -g 1 -f 20

Something should be displayed on the OLED within a couple of seconds. If nothing is playing a clock screen will display, and this turns to the screen with the spectrum display when music is playing. See this video: https://youtu.be/Jl63Dt5A0Jo

Adrian.

Korniman-the-Rooky, 2020-02-26:
Dear Adrian,
It seems it will not work ahhhhhhhhh.
I used a new image with Volumio V2.712 on my raspberry and did all new with your new code, without success.
After that a coleague of mine from my company gave me his raspi and we both (he is much deeper into linux etc.) made a very simple installation on his Raspi 4 to carefully check that a pin of my Raspi is not damaged or somewhat else.
We only used Raspi, OLED, USB Stick for the Music, USB Soundcard. We also test it with the Raspi Audiojack without external soundcard

We installed "sudo apt install raspi-config" as before to install the configuration menu and enabled SPI interface, but we also tried out your style to enable the SPI:

nano /boot/userconfig.txt

# Add your custom config.txt options to this file, which will be preserved during updates
dtparam=spi=on

we also configured the file as before:
sudo nano /volumio/app/plugins/music_service/mpd/mpd.conf.tmpl

```
# Character Encoding ##########################################################
filesystem_charset              "UTF-8"
id3v1_encoding                  "UTF-8"
###############################################################################

# MPD CAVA #########################################
audio_output {
        type            "fifo"
        name            "mpd_oled_FIFO"
        path            "/tmp/mpd_oled_fifo"
        format          "44100:16:2"
}

```

then we played a simple mp3
we also edited 
nano mpd_oled.service with the correct OLED o- type

No Function - black display. We tried it with SPI Mode for the following displays SSD1106 and SSD 1309.

we also play with execute "sudo bash install.sh" and "sudo systemctl start mpd_oled"

No success

I there something more which we could test. If you want we could upload the image from the SD-Card
I am devestated that I am not able to reproduce your installation

Kind regards Benjamin

antiprism, 2020-02-26:
Hi Benjamin

I am running Volumio 2.703, from about a week ago.

You could check that SPI is available
```
volumio@volumio5:~$ ls /dev/spidev*
/dev/spidev0.0  /dev/spidev0.1
```
What is mpd_oled doing when it is running and the screen is dark? Normally, it should be busier when music is playing on Volumio through MPD, e.g. a radio station (but only if the copy of the audio has been configured correctly).
Run 'top' when no music is playing. For me mpd_oled=6.9% CPU, cava=0.7% CPU
```
  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND      
  886 volumio   20   0  197180  93116  24780 S  8.2 19.6   1:40.08 node         
 2239 root      20   0   32900   7584   6624 S  6.9  1.6   0:32.62 mpd_oled     
  145 root      20   0    8364   2788   2488 S  3.3  0.6   0:18.47 systemd-jou+ 
  297 root      20   0       0      0      0 S  1.0  0.0   0:02.91 kworker/u2:+ 
12396 volumio   20   0    5016   2180   1816 R  1.0  0.5   0:00.22 top          
    7 root      20   0       0      0      0 S  0.7  0.0   0:03.80 ksoftirqd/0  
  530 mpd       20   0  163188  32228  21232 S  0.7  6.8   0:10.61 mpd          
 1003 volumio   20   0   89336   7708   6760 S  0.7  1.6   0:04.90 upmpdcli     
 1004 shairpo+  20   0   87376   7072   6232 S  0.7  1.5   0:12.89 shairport-s+ 
 2241 root      20   0   15316   4108   2780 S  0.7  0.9   0:05.09 cava         
```

Run top when music is playing through a Volumio radio station. For me mpd_ole=13.8% CPU, cava=6.1% CPU
```
  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND      
  530 mpd       20   0  171892  34828  21232 S 18.0  7.3   0:29.63 mpd          
 2239 root      20   0   32900   7584   6624 R 13.8  1.6   0:55.88 mpd_oled     
  886 volumio   20   0  200252  83624  24780 S  8.4 17.6   2:00.15 node         
 2241 root      20   0   15316   4108   2780 S  6.1  0.9   0:12.21 cava         
```
This is my OLED (1.3", SSH1106)
https://www.ebay.es/itm/0-96-1-3-SPI-Serial-128X64-OLED-LCD-Display-Screen-Module-Arduino-UNO-R3/152315345796?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649

I cloned the mpd_oled repository from fresh, built the program, and ran it with
```sudo ./mpd_oled -o 7 -b 10 -g 1 -f 20```

Here are a couple of photos of my setup (the extra card is an I2S sound card). If you are able to follow the wires they should be connected the same was yours.

[image]

[image]

Adrian.

Korniman-the-Rooky, 2020-02-26:
Dear Adrian, thank you for to much help ;-)

Here are the results from my pi currently with the onboard sound card and fresh system

```
volumio@volumio:~$ ls /dev/spidev*
/dev/spidev0.0  /dev/spidev0.1
volumio@volumio:~$
```

with No Music and no"sudo ./mpd_oled -o 1 -b 10 -g 1 -f 20"

```
PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
3504 volumio   20   0    4328   3224   2020 S   0.7  0.1   0:01.08 htop
10 root      20   0       0      0      0 I   0.3  0.0   0:00.32 rcu_sched
1163 shairpo+  20   0   91484   6464   5688 S   0.3  0.2   0:04.05 shairport-sync

```
with No Music and execute "sudo ./mpd_oled -o 1 -b 10 -g 1 -f 20"

```
 PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
 4063 root      20   0   14232   5140   4492 R  **99.9**  0.1   1:09.88 mpd_oled
  946 volumio   20   0  213708  94068  24644 S   0.7  2.3   0:13.45 node
  642 mpd       20   0  190988  43112  26388 S   0.3  1.1   0:06.08 mpd
 1163 shairpo+  20   0   91484   6464   5688 S   0.3  0.2   0:04.40 shairport-sync
 1255 volumio   20   0  110852   7780   6712 S   0.3  0.2   0:02.71 upmpdcli
 4201 volumio   20   0    3580   2188   1756 R   0.3  0.1   0:00.04 top
```

with Music on and execute "sudo ./mpd_oled -o 1 -b 10 -g 1 -f 20"

 ```
PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
 4063 root      20   0   14232   5140   4492 R **100.0**  0.1   3:52.05 mpd_oled
  642 mpd       20   0  190988  45312  26452 S   2.3  1.1   0:06.98 mpd
 3504 volumio   20   0    4328   3224   2020 S   0.7  0.1   0:02.71 htop
 4201 volumio   20   0    3580   2188   1756 R   0.7  0.1   0:00.58 top
   62 root       1 -19       0      0      0 S   0.3  0.0   0:00.36 vchiq-slot/0
  946 volumio   20   0  214732  95440  24644 S   0.3  2.4   0:13.90 node
 1163 shairpo+  20   0   91484   6464   5688 S   0.3  0.2   0:04.75 shairport-sync
 1255 volumio   20   0  110852   7792   6712 S   0.3  0.2   0:03.06 upmpdcli
 4182 volumio   20   0    9380   3480   2828 S   0.3  0.1   0:00.13 sshd
    1 root      20   0   24208   4136   2740 S   0.0  0.1   0:04.42 systemd
    2 root      20   0       0      0      0 S   0.0  0.0   0:00.00 kthreadd
    3 root       0 -20       0      0      0 I   0.0  0.0   0:00.00 rcu_gp
    4 root       0 -20       0      0      0 I   0.0  0.0   0:00.00 rcu_par_gp

```

We did find "mpd_oled" with 100% CPU. CPU is realy warm
We did not find "cava"
but installation was fine

what should I check next?

Kind Regards Benjamin

antiprism, 2020-02-26:
Hi Benjamin

I don't know if it will help, but I have updated the library code that handles the I2C and SPI communication. The version I was using seems to be several years old, and the library has received a lot of updates since then (maybe the old version has compatibility issues with newer Pis) . I recommend cloning a fresh copy of the mpd_oled repository. Then, run "PLAYER=VOLUMIO make" and then try running a test command with the new mpd_oled program.

Regarding your results, mpd_oled shouldn't run at 100%. It looks like it is encountering some kind of problem before it launches 'cava', most likely when trying to initialise the display. If you can get someone who knows C or C++ to help you track down where in the code the issue is occurring, I might be able to suggest something.

Adrian.

Maschine2501, 2020-02-26:
Hi Adrian!
Im pleased to see here some changes!
Have you integrated the ssd1322 support also?
I think my issue with it, maybe, is a result of the changes (you mentioned it above?)...
I realy want to give it a try, but it never displayed anything at all.

Meanwhile I've investigated in the code "Volumio-OledUi" and have made some steps forward in understanding the code.
Here's the link: github.com/Maschine2501/Volumio-OledUI

Greetings!

antiprism, 2020-02-26:
Hi

I have only added support for SPI screens with SSD1306/9 and SSH1106 chips, and if the SSD1322 needs specific controller code then the newer version of the SPI library probably won't help.

Volumio-OledUI looks like a cool project!

Adrian.

Korniman-the-Rooky, 2020-02-26:
Dear Adrian, thank you I will give it a hopefully trail tomorrow.
By the way I checked your setup on your foto. OLED setup is the same as mine. I thought you had also the Raspi 4 in the meanwhile. So maybe it could be that there are some differences in the layout from Raspi zero to 4 which causes the troubles. If your update will not bring efforts I will try to find someone with C++. If it would be easier for you I could also send you a remote link. But this is up to you.
I installed also the TOPH app which brings clearer results than TOP in the task load function. There I searched for wording CAVA and got marked that task "shareport" has something to do with CAVA
Good night
Benjamin

Korniman-the-Rooky, 2020-02-27:
Dear Adrian, you are my man of the hour!
All works fine now, updating the library code that handles the I2C and SPI was the goal.
Now I tested it with variable displays and all works fine. Even the original configuration works

Current Setup is 
Raspberry Pi 4
Khadas Tone Board
Remot Pi Board
Real Time Clock
1.54 OLED with SSD1309 SPI 

All works perfect

Thank you so much for your realy good support
Kind Regards Benjamin

antiprism, 2020-02-27:
Hi Benjamin

That is great! I am really pleased your displays are now working.

Thanks for all your feedback and patience while troubleshooting the issue!

Adrian.

antiprism, 2020-02-27:
Closed by: https://github.com/antiprism/mpd_oled/commit/410fcba2ce5349915d0df100909a294eeaa9925f and https://github.com/antiprism/mpd_oled/commit/d9644140ac99f76e1ebb87530e8cf117f5aa4e77

closed 2020-02-27 by antiprism