antiprism/mpd_oled #37: Delay and Refresh

fabio323ti, 2020-04-12:
i'd follow all steps and working almost fime...
so.. i have a slow refresh on oled display (ssd1309) , expecially if i use GPIO2 and GPIO3
after moved to GPIO 5 and GPIO6 (overlay of i2c) run more smooth but not perfectly..
and another issue is delay of spectrum from music... display delay about 2 sec.

my config is Moode audio with RP 3b+ without Dac board

antiprism, 2020-04-12:
Hi

The slow refresh and delay are probably the same isue, of sending more data than the i2c bus can handle.

Check that you have set the bus speed fairly high, e.g.
```dtparam=i2c_arm_baudrate=800000```
Then see if low values of bars and FPS produce a synced display, e.g. ```-b 10 -f 10```. If they do, try to increase -f in steps up to 25, and if you can do this then, with ```-f 25```, try to increase -b in steps up to some higher value you like.

Adrian.

fabio323ti, 2020-04-12:
thanks Adrian, already tested your mod, results good but not realtime, have high latency, 
so. i guess is i2c issue, will try spi connection.
So.. does the spectrum is "cava settings" dependant? or uses fixed setup in mpd bundle?
thanks

fabio323ti, 2020-04-13:
Solved, resinstall all from zero and repeat steps.. 
so dtparam=i2c_arm_baudrate=800000 is mandatory for smooth results...

closed 2020-04-13 by fabio323ti