antiprism/mpd_oled #39: New build on upto date pi4 creates random jitter on bar display

wildestpixel, 2020-04-17:
Firstly - I recreated a new build as old sd card was getting a bit messy and introducing audio dropout on mpd. This said the output of bars was clean and perfect.

Move forward to this week, reinstalled everything on a fresh build and getting this jitter effect. Cava seems to build cleanly, output of make on mpd_oled : 

```
pi@quattro:~/mpd_oled $ PLAYER=MPD LDLIBS="-li2c" make
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o main.o main.cpp
In file included from /usr/include/c++/8/vector:69,
                 from display_info.h:29,
                 from display.h:32,
                 from main.cpp:25:
/usr/include/c++/8/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const double&}; _Tp = double; _Alloc = std::allocator<double>]’:
/usr/include/c++/8/bits/vector.tcc:413:7: note: parameter passing for argument of type ‘std::vector<double>::iterator’ {aka ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’} changed in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/vector:64,
                 from display_info.h:29,
                 from display.h:32,
                 from main.cpp:25:
/usr/include/c++/8/bits/stl_vector.h: In constructor ‘OledOpts::OledOpts()’:
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h: In member function ‘virtual void OledOpts::process_command_line(int, char**)’:
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o timer.o timer.cpp
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o status.o status.cpp
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o status_msg.o status_msg.cpp
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o utils.o utils.cpp
In file included from /usr/include/c++/8/vector:69,
                 from utils.h:38,
                 from utils.cpp:29:
/usr/include/c++/8/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const double&}; _Tp = double; _Alloc = std::allocator<double>]’:
/usr/include/c++/8/bits/vector.tcc:413:7: note: parameter passing for argument of type ‘std::vector<double>::iterator’ {aka ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’} changed in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/vector:64,
                 from utils.h:38,
                 from utils.cpp:29:
/usr/include/c++/8/bits/stl_vector.h: In function ‘Status read_double_list(char*, std::vector<double>&, int, const char*)’:
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_vector.h: In function ‘Status read_double_list(std::vector<char*>&, std::vector<double>&)’:
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o display.o display.cpp
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o programopts.o programopts.cpp
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o ultragetopt.o ultragetopt.cpp
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o ArduiPi_OLED.o ArduiPi_OLED.cpp
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o Adafruit_GFX.o Adafruit_GFX.cpp
cc  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o bcm2835.o bcm2835.c
bcm2835.c: In function ‘bcm2835_gpio_pad’:
bcm2835.c:487:3: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
   if (bcm2835_pads == MAP_FAILED)
   ^~
bcm2835.c:490:5: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
     volatile uint32_t* paddr = bcm2835_pads + BCM2835_PADS_GPIO_0_27/4 + group;
     ^~~~~~~~
bcm2835.c: In function ‘bcm2835_gpio_set_pad’:
bcm2835.c:500:3: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
   if (bcm2835_pads == MAP_FAILED)
   ^~
bcm2835.c:503:5: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
     volatile uint32_t* paddr = bcm2835_pads + BCM2835_PADS_GPIO_0_27/4 + group;
     ^~~~~~~~
cc  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o bcm2835_i2c.o bcm2835_i2c.c
bcm2835_i2c.c: In function ‘bcm2835_i2c_alt_write’:
bcm2835_i2c.c:89:14: warning: implicit declaration of function ‘i2c_smbus_write_byte_data’ [-Wimplicit-function-declaration]
     reason = i2c_smbus_write_byte_data(i2c_fd, buf[0], buf[1]);
              ^~~~~~~~~~~~~~~~~~~~~~~~~
bcm2835_i2c.c:92:14: warning: implicit declaration of function ‘i2c_smbus_write_word_data’ [-Wimplicit-function-declaration]
     reason = i2c_smbus_write_word_data(i2c_fd, buf[0], (buf[2] << 8) | buf[1]);
              ^~~~~~~~~~~~~~~~~~~~~~~~~
bcm2835_i2c.c:95:14: warning: implicit declaration of function ‘i2c_smbus_write_i2c_block_data’ [-Wimplicit-function-declaration]
     reason = i2c_smbus_write_i2c_block_data(i2c_fd, buf[0], len - 1,
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
g++  -o mpd_oled main.o timer.o status.o status_msg.o utils.o display.o programopts.o ultragetopt.o ArduiPi_OLED.o Adafruit_GFX.o bcm2835.o bcm2835_i2c.o -li2c -lmpdclient -lpthread
```

Don't know if the build characteristics have any bearing on this jitter, or whether something is carrying forward from cava. Any variables you need - point me in the direction to give them to you. The jitters are bars that dont represent audio but flick fully on (full amplitude) anywhere on the spectrum and generally in quiet spoken word (radio broadcast). The rest of the display is perfect. Tested on an option 6 SH1106 and option 3 Adafruit 1.3" and both the same - its not the displays and as I say - not evident on old SD card - all same hardware.

I have a video of the issue I can mail drop to you.

antiprism, 2020-04-17:
Hi

Thanks for reporting the issue. I have been able to replicate it on a Pi Zero.

The mpd_oled warnings are all normal. I updated to the latest mpd_oled and all was fine. I updated to the latest cava and the issue appeared. I will raise this with the cava project.

In the meantime, you could use an older version of cava, e.g.
```
git clone https://github.com/karlstav/cava
cd cava
git checkout f60c79cab
./autogen.sh
./configure
make
sudo make install
```
Adrian.

wildestpixel, 2020-04-17:
Thanks for this – that’s confirmed as resolving by regressing the version of cava.

 

Sorry to have landed this at your doorstep!

antiprism, 2020-04-20:
Cava has been updated with a fix for this. Closing issue.

closed 2020-04-20 by antiprism