antiprism/mpd_oled #40: arm64

antonellocaroli, 2020-04-22:
I'm trying to compile on a gentoo aarh64 (arm64) system.

but it gives me this error, does it depend on the 64bit system?

PLAYER=MPD make
c++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o main.o main.cpp
c++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o timer.o timer.cpp
c++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o status.o status.cpp
c++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o status_msg.o status_msg.cpp
c++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o utils.o utils.cpp
c++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o display.o display.cpp
c++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o programopts.o programopts.cpp
c++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o ultragetopt.o ultragetopt.cpp
c++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o ArduiPi_OLED.o ArduiPi_OLED.cpp
c++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o Adafruit_GFX.o Adafruit_GFX.cpp
cc  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o bcm2835.o bcm2835.c
bcm2835.c: In function ‘bcm2835_gpio_pad’:
bcm2835.c:487:3: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  487 |   if (bcm2835_pads == MAP_FAILED)
      |   ^~
bcm2835.c:490:5: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  490 |     volatile uint32_t* paddr = bcm2835_pads + BCM2835_PADS_GPIO_0_27/4 + group;
      |     ^~~~~~~~
bcm2835.c: In function ‘bcm2835_gpio_set_pad’:
bcm2835.c:500:3: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  500 |   if (bcm2835_pads == MAP_FAILED)
      |   ^~
bcm2835.c:503:5: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  503 |     volatile uint32_t* paddr = bcm2835_pads + BCM2835_PADS_GPIO_0_27/4 + group;
      |     ^~~~~~~~
cc  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DMPD  -c -o bcm2835_i2c.o bcm2835_i2c.c
bcm2835_i2c.c: In function ‘bcm2835_i2c_alt_write’:
bcm2835_i2c.c:89:14: warning: implicit declaration of function ‘i2c_smbus_write_byte_data’ [-Wimplicit-function-declaration]
   89 |     reason = i2c_smbus_write_byte_data(i2c_fd, buf[0], buf[1]);
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~
bcm2835_i2c.c:92:14: warning: implicit declaration of function ‘i2c_smbus_write_word_data’ [-Wimplicit-function-declaration]
   92 |     reason = i2c_smbus_write_word_data(i2c_fd, buf[0], (buf[2] << 8) | buf[1]);
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~
bcm2835_i2c.c:95:14: warning: implicit declaration of function ‘i2c_smbus_write_i2c_block_data’ [-Wimplicit-function-declaration]
   95 |     reason = i2c_smbus_write_i2c_block_data(i2c_fd, buf[0], len - 1,
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c++  -o mpd_oled main.o timer.o status.o status_msg.o utils.o display.o programopts.o ultragetopt.o ArduiPi_OLED.o Adafruit_GFX.o bcm2835.o bcm2835_i2c.o  -lmpdclient -lpthread 
/usr/lib/gcc/aarch64-unknown-linux-gnu/9.3.0/../../../../aarch64-unknown-linux-gnu/bin/ld: bcm2835_i2c.o: in function `bcm2835_i2c_alt_write':
bcm2835_i2c.c:(.text+0x13c): undefined reference to `i2c_smbus_write_i2c_block_data'
/usr/lib/gcc/aarch64-unknown-linux-gnu/9.3.0/../../../../aarch64-unknown-linux-gnu/bin/ld: bcm2835_i2c.c:(.text+0x160): undefined reference to `i2c_smbus_write_word_data'
/usr/lib/gcc/aarch64-unknown-linux-gnu/9.3.0/../../../../aarch64-unknown-linux-gnu/bin/ld: bcm2835_i2c.c:(.text+0x170): undefined reference to `i2c_smbus_write_byte_data'
collect2: error: ld returned 1 exit status
make: *** [Makefile:30: mpd_oled] Error 1

antiprism, 2020-04-22:
Hi

Thanks for reporting the issue.

I include bcm2835.c exactly as it is distributed by the bcm2835 library project, and I have left the warnings in case they are ever relevant.

The implicit declaration warnings for i2c_smbus* are due to not including a suitable header, and can be ignored.

The error is because on your system these  i2c_smbus* functions are in a separate libray. You should be able to indicate this and complete the build with

```PLAYER=MPD LDLIBS="-li2c" make```

Adrian.

antonellocaroli, 2020-04-22:
Thanks! It worked!

antiprism, 2020-04-22:
Great! Thanks for letting me know.

closed 2020-04-22 by antiprism