antiprism/mpd_oled #41: make error - fiel mpd/client.h missing

pr-boh, 2020-04-26:
Hi, I try to install mpd_oled on a raspberry pi with mpd running (internet-radio).
When starting make with 'PLAYER=MPD make', I get the following error:
In file included from display_info.h:28:0,
                 from display.h:32,
                 from main.cpp:25:
status.h:28:24: fatal error: mpd/client.h: Datei oder Verzeichnis nicht gefunden
 #include <mpd/client.h>
compilation terminated.

I have no idea, where to find the missing file?

antiprism, 2020-04-26:
It looks like you might not have installed a libmpdclient development package for your system. Are you running Raspbian and did you install all the dependencies in the Volumio install instructions?
```
sudo apt install build-essential git-core autoconf make libtool libi2c-dev i2c-tools lm-sensors libcurl4-openssl-dev libmpdclient-dev libjsoncpp-dev
```
Adrian.

pr-boh, 2020-04-26:
Hi Adrian, i forgot ‚-sudo apt update- and not all of the dependencies installed.  
With ‚sudo apt update‘ and sudo apt install xxx, all dependencies installed and make is running. 
Thank you very much for your fast help, that solved the problem. 

Paul

„Wissen ist das einzige Gut, das sich vermehrt, wenn man es teilt.“
   Marie Freifrau von Ebner-Eschenbach (1830 – 1916)

antiprism, 2020-04-26:
Great! Thanks for letting me know.

closed 2020-04-26 by antiprism