antiprism/mpd_oled #42: Cava slow to respond

Dave0616, 2020-05-14:
It took me quite a while to get this working but I was able to, but when I click play the graphic EQ starts up and seems to be working (sometimes I have around a 9 second delay before it starts), click pause, bars keep moving for about 12 seconds... I have tried 2 versions of volumio and both do it (latest and 2.413)

do you have any idea what might be causing this?

antiprism, 2020-05-14:
Hi

Try the steps given in https://github.com/antiprism/mpd_oled/issues/37

Adrian.

Dave0616, 2020-05-14:
I wonder if this is an issue with my oled? I have tried changing the baus rate a few times, im not sure if its actually changing but I have tried

dtparam=i2c_arm_baudrate=400000
dtparam=i2c_arm_baudrate=800000
dtparam=i2c_arm_baudrate=40000

and I think 

dtparam=i2c_arm_baudrate=100000

all have no effect on it

also this is the only command that gets my oled to start

./mpd_oled -o3

this either doesnt work or causes it to go crazy

./mpd_oled -o 6 -b 10 -g 1 -f 20

Thank you for your help!

antiprism, 2020-05-14:
Did you try a command with reduced -b and -f?  E.g.

```./mpd_oled -o 3 -b 10 -g 1 -f 10```

(with, say, 'dtparam=i2c_arm_baudrate=400000')

Dave0616, 2020-05-14:
changed the baudrate and the command, anything I put in causes the screen to freak out!
I ordered what I think is the screen you used and ill try that

https://www.amazon.com/gp/product/B01MRR4LVE/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

this is the screen im using

https://www.amazon.com/gp/product/B01HHOETIA/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

here are the commands im using and the sequence, the bars dont even show themselves until I go into the volumio settings and start clicking save on stuff (im not sure it its the volume or the output I need to click save on im still working that out) I need to be able to redo this a few times so I keep track of what im doing

#Enable SSH on volumio
#go to your volumio web interface page and add "/dev" after its address (see next line)

http:// IP Address /dev/ (example http://10.0.0.1/dev/)

#Putty into volumio
#Username: volumio
#Password: volumio

sudo apt-get update
sudo apt-get install git-core autoconf make libtool libfftw3-dev libasound2-dev
git clone https://github.com/karlstav/cava
cd cava
./autogen.sh
./configure
make
sudo make install

#reboot pi (sudo reboot)
#Log back in with putty

dtparam=i2c_arm_baudrate=800000  

#reboot pi (sudo reboot)
#Log back in with putty

sudo nano /boot/config.txt (older volumio)
sudo nano /boot/userconfig.txt (newer volumio)

dtparam=spi=on

sudo nano /volumio/app/plugins/music_service/mpd/mpd.conf.tmpl

#add this line at the end

audio_output {
        type            "fifo"
        name            "mpd_oled_FIFO"
        path            "/tmp/mpd_oled_fifo"
        format          "44100:16:2"
}

#ctrl+x
#y
#enter

sudo dpkg-reconfigure tzdata

sudo reboot

#Log back in with putty

sudo apt install build-essential git-core autoconf make libtool libi2c-dev i2c-tools lm-sensors libcurl4-openssl-dev libmpdclient-dev libjsoncpp-dev

git clone https://github.com/antiprism/mpd_oled

cd mpd_oled
PLAYER=VOLUMIO make

sudo ./mpd_oled -o3

#screen should come on

sudo nano mpd_oled.service

#edit [Service] to the following
#ExecStart=/usr/local/bin/mpd_oled -o3

#ctrl+x
#y
#enter

sudo bash install.sh

sudo systemctl start mpd_oled

#you may have to reboot after that?

#go to volumio "settings" chose "audio output" go down to "Volume Options" click save, reboot

antiprism, 2020-05-14:
Your OLED should be fine. The oled option is -o3, which is what you are correctly using.

As you have run the install.sh script then mpd_oled might already be running. If this is the case then first stop and disable the installed version from running before running the test commands

```
sudo systemctl stop mpd_oled
sudo systemctl disable mpd_oled
```
The screen should now be blank and you can try running the test command

```
./mpd_oled -o 3 -b 10 -g 1 -f 10
```
Reviewing your command list, have you added the following line to  /boot/userconfig.txt (or /boot/config.txt )?
```
dtparam=i2c_arm_baudrate=400000
```
If not, this would be the cause of the delay (and if it works, try increasing the value to 800000)

Otherwise, try reveiwing all your connections to make sure that the wires and soldering is good (also the wires may need to be 20 cm or shorter).

Dave0616, 2020-05-14:
that seems to work better!

still a bit off but much better, ill try changing the code  a bit more to see if I can get it to respond faster

the /boot/userconfig.txt codes I have tried on both (im using 2 versions of volumio in testing) but it doesnt seem to have an effect even if I dont edit that file but with all the attempts I have made I might be mistaken! Ill reply if I can find a good setting, and ty again

antiprism, 2020-07-03:
Closing as no further updates. Please feel free to reopen.

closed 2020-07-03 by antiprism