antiprism/mpd_oled #43: Support for SSD1306 6-Pin SPI OLED Display

rradar, 2020-05-24:
Hi @antiprism 

thank's for your work already.

I need to tell that I some how stuck before I got things running. I made a fresh install of moode 6.5.2 to have a clean start but still my display doesn't light up. I followed your setup instructions exactly and there were no errors at all which leads me to the last possible reason: my display is not supported?

I have this type of display
[image]

Found this link online (my version is only single colored): https://www.rhydolabz.com/displays-c-88/096-oled-bicolor-display-spii2c-128x64-6-pin-blueyellow-p-2262.html

The thing with this display (which supported I2C and SPI) is that it is shipped as SPI (4SPI?). I have no equipment to change it to I2C and also want to profit from the higher refresh rate for SPI.

The display has only 6 pins and is missing CS pin. Otherwise I wired exactly like you [describe](https://github.com/antiprism/mpd_oled/blob/master/wiring_spi.png).

I tried various display settings (most importantly `-o 1` and `-o 7` I guess).

Is there any hope to get this display running in "stock" setting?

antiprism, 2020-05-24:
Hi

Your OLED should be supported. The description says the controller is SSD1306, but in the Q&A it says SH1106, so -o1 or -o7 should work, depending on what it really is.

I notice there is an error in the mpd_oled Moode instructions for enabling SPI (which were copy pasted from the Volumio instruction, I will fix this).

  https://github.com/antiprism/mpd_oled/blob/master/INSTALL_MOODE.md#spi

On Moode, you have to edit /boot/config.txt , so if you edited /boot/userconfig.txt then this might be the cause of the problem.

I don't think the lack of the CS pin is a problem.

Adrian.

rradar, 2020-05-24:
> On Moode, you have to edit /boot/config.txt

I have this line in it:
`dtparam=spi=on`

If I run `sudo ./mpd_oled -o 1 -b 10 -g 1 -f 20` or `sudo ./mpd_oled -o 7 -b 10 -g 1 -f 20`

I get: `mpd_oled: error: could not initialise OLED`

Any ideas for debugging? I tried already to swap CLK & MOSI pins to be sure they are not wrong

antiprism, 2020-05-25:
I am not sure what the problem is. It looks like you are doing everything correctly.

I am still on Moode 6.4.2 here, so I'll check the mpd_oled install instructions using an SPI OLED on a fresh copy of the latest Moode in case this is an issue with mpd_oled. I'll post an update later.

antiprism, 2020-05-25:
This is an issue with mpd_oled. I have just pushed a fix: https://github.com/antiprism/mpd_oled/commit/34831a50639b695aeda82e46da6670e271ef5ffa

There were two issues (!). Previous changes to the I2C initialisation code were stopping the SPI  initialisation code from being called, also, the wiring diagram was not using defaults for the RST and DC connections.

If you download the lastest mpd_oled, and follow the updated SPI wiring diagram
https://raw.githubusercontent.com/antiprism/mpd_oled/master/wiring_spi.png
then your display will hopefully now work.

Thanks for reporting the issue!

rradar, 2020-05-25:
Thank's for your updated code @antiprism 

I pushed the new version and went for the new wiring.

Sadly I still don't have any pixel on the display lightning up - but at least the old error message (`mpd_oled: error: could not initialise OLED`) is completely gone.

Is there any verbose mode so that I could see where it could stuck? I will try other cables in the meantime... sometimes it's only the wiring - you never know!

antiprism, 2020-05-25:
I think your display might be broken, caused by the screen being damaged in the bottom left corner (and bottom right corner?). The following two displays are damaged in the bottom left corner and have similar issues to what you are seeing
http://moodeaudio.org/forum/showthread.php?tid=155&pid=3638#pid3638
https://forums.adafruit.com/viewtopic.php?f=47&t=34422

rradar, 2020-05-25:
> I think your display might be broken, caused by the screen being damaged in the bottom left corner (and bottom right corner?).

You are mostly right. These tiny oled displays are really fragile on this corners. I think that 4 of my 5 have are broken but for now I just had issues with one of them (some dead lines).

Any how I'm closing this. Because it's solved! Thank's for your help, fixes and this great mpd_oled project! :star2:

closed 2020-05-25 by rradar