antiprism/mpd_oled #44: Is MPD_Oled compatible to Volumio snapcast plugin?

Korniman-the-Rooky, 2020-06-29:
Dear Adrian, is MPD_Oled compatible to Volumio snapcast plugin for multi room solution?
Currently my Raspi pops up failure while activating the plugin and later on the volumio player has malfunction.
Thanx for your support
Benjamin

antiprism, 2020-06-29:
Hi Benjamin

Did you set up the audio copy for mpd_oled like ths?
https://github.com/antiprism/mpd_oled/blob/master/INSTALL_VOLUMIO.md#configure-copy-of-audio

What is the error you get when you activate the snapcast plugin (it would help if you could include the exact text, copy and paste if possible). What happens when Volumio later malfunctions (what specifically is not working)?

Adrian.

Korniman-the-Rooky, 2020-07-03:
Dear Adrian, I made a clean New Image with pure Volumio and activated only
snapcast plugin. It also did not work. So sorry for the trouble. I will
first try to fix the basics.
Thanx
Benjamin

Mit freundlichen Grüßen
Benjamin Korntheuer

antiprism, 2020-07-03:
Hi Benjamin

No problem. Thanks for letting me know!

Adrian.

closed 2020-07-03 by antiprism