antiprism/mpd_oled #45: Rune Audio enabling mpd_oled on boot

Ephesian, 2020-08-27:
to get the mpd_oled to be enabled on power on, rather than putting an entry for i2c-dev in /etc/modules you need to create a file in /etc/modules-load.d.  eg. i2c.conf, with the entry i2c-dev

antiprism, 2020-08-28:
Hi Ephesian

Thanks for the feedback. Looking at the Archlinux documentation I can see that this is the only method proposed. Do you think that this should be the only method mentioned in the mpd_oled install instructions, as it would work for all official and forum versions of RuneAudio? If this is the case, I could change that part of the instructions to read:

I created the file /etc/modules-load.d/i2c.conf and added the line ```i2c-dev```.

Adrian

antiprism, 2022-11-05:
Closing, as no feedback.

closed 2022-11-05 by antiprism