antiprism/mpd_oled #46: CAVA make fails

guanilee, 2020-09-20:
Hello,
I'm getting an error as I try to make CAVA.
I'm building this up on top of the Volumio installed on a Pi zero W.
With very little knowledge about programming, I'm at a loss how to solve this.
What should I do?

Message I get is as follows:
===========================================================================
cava.c: In function 'main':
cava.c:399:36: error: 'M_PI' undeclared (first use in this function)
                 0.5 * (1 - cos(2 * M_PI * i / (audio.FFTbassbufferSize - 1)));
                                    ^
cava.c:399:36: note: each undeclared identifier is reported only once for each function it appears in
cava.c: At top level:
cc1: error: unrecognized command line option "-Wno-unknown-warning-option" [-Werror]
cc1: all warnings being treated as errors
Makefile:588: recipe for target 'cava-cava.o' failed
make[1]: *** [cava-cava.o] Error 1
make[1]: Leaving directory '/home/volumio/cava'
Makefile:792: recipe for target 'install-recursive' failed
make: *** [install-recursive] Error 1

antiprism, 2020-09-20:
Hi

Thank you for reporting this issue. I was able to reproduce it.

To work around the issue, change to the cava directory and complete the cava build and installation with
```
make clean
CPPFLAGS=-DM_PI=3.1415926535897932385 ./configure
make
sudo make install
```
I'll let the cava project know about the issue.

Adrian.

guanilee, 2020-09-21:
[image]

Thank you so much!
I got it working now, and it's exciting to see it.
Thank you so much for this wonderful project. :)

closed 2020-09-21 by guanilee