antiprism/mpd_oled #47: MPD Info without CAVA?

catalinescu-sudo, 2020-10-21:
Hello!

Is it possible to install the service as it is without the spectrum visualiser? On rpi 0 w the installing of mpd_oled makes the device respond laggy, and I believe is from the spectrum part.
I think the older versions were without the spectrum. I only need song info and clock screensaver when nothing is playing. It's doable?

Thank you for your support on volumio community

Best regards,
Catalin

antiprism, 2020-10-21:
Hi Catalin

The spectrum display has always been included. If you want to remove it then you will need to change the code, It is probably sufficient to have have mpd_oled read the audio from a FIFO that doesn't exist ('mpd_oled -c fifo,dummy') and modify
the draw_spect_display() function in main.c to draw the display elements that you want.

Alternatively, more bars and a higher framerate require more processing, so try running running with e.g. 'mpd_oled -b 10 -f 15', and see if the player is more responsive.

Also, try running 'top' to see how much CPU the different proceses are using when the player is unresponsive.

Adrian.

supercrab, 2021-02-15:
@catalinescu-sudo I had `mpd_oled` running on my Pi 0 initially.  If you want a snappier system, try a different Pi model like RPi 3, it's much more responsive and might be worth the extra £££.

catalinescu-sudo, 2021-02-15:
Yes indeed but this is my actual setup based on the allready designed and
printed case.