antiprism/mpd_oled #48: Compilation failing and compiler warning and state not defined

supercrab, 2020-11-19:
Weird one: I'm receiving this warning whilst compiling **warning: 'auto' changes meaning in C++11; please remove it [-Wc++0x-compat]**

and the build fails due to the use of auto keyword for defining state in ```main.cpp:444```

This was building fine last week :|

```root@volumio:/home/volumio/mpd_oled# PLAYER=VOLUMIO make
g++  -W -Wall -Wno-unused-variable -Wno-unused-parameter -Wno-strict-aliasing -Ofast -DVOLUMIO  -c -o main.o main.cpp
main.cpp: In function 'void draw_display(ArduiPi_OLED&, const display_info&)':
main.cpp:444:3: warning: 'auto' changes meaning in C++11; please remove it [-Wc++0x-compat]
   auto state = disp_info.status.get_state();
   ^
main.cpp:444:8: error: 'state' does not name a type
   auto state = disp_info.status.get_state();
        ^
main.cpp:445:7: error: 'state' was not declared in this scope
   if (state == MPD_STATE_UNKNOWN || state == MPD_STATE_STOP ```

supercrab, 2020-11-19:
I can see you've just fixed it ```mpd_state state```.  I've made the changed manually and fixed it :)

closed 2020-11-19 by supercrab

antiprism, 2020-11-19:
Hi Mr Mase

Thanks for reporting this. It was introduced with the previous commit. You beat me to replying!

Adrian.

supercrab, 2020-11-19:
Hey Adrian!

Glad to hear it's all sorted.  Take it easy :)

Mase