antiprism/mpd_oled #49: New feature implemented: disable equaliser when paused

supercrab, 2020-11-30:
Hi Adrian,

I've implemented a new feature whereby you can disable the equaliser when the player is paused with the `-e` parameter  This is useful if you're using the mic input as the audio source of the equaliser with `-c alsa,hw:5,0`   When the player is paused, CAVA takes the noise from the input and slowly normalises the data so that it fits the entire graph, even though there is nothing playing.  I simply clear the array that holds the spectrum data when the player is paused.
`

supercrab, 2020-11-30:
`/*
   Copyright (c) 2018, Adrian Rossiter

   Antiprism - http://www.antiprism.com

   Permission is hereby granted, free of charge, to any person obtaining a
   copy of this software and associated documentation files (the "Software"),
   to deal in the Software without restriction, including without limitation
   the rights to use, copy, modify, merge, publish, distribute, sublicense,
   and/or sell copies of the Software, and to permit persons to whom the
   Software is furnished to do so, subject to the following conditions:

      The above copyright notice and this permission notice shall be included
      in all copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
  IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
  AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
  FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS
  IN THE SOFTWARE.
*/

#include "display.h"
#include "display_info.h"
#include "timer.h"
#include "programopts.h"
#include "utils.h"

#include <stdio.h>
#include <stdlib.h>
#include <signal.h>
#include <unistd.h>
#include <pthread.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <locale.h>
#include <errno.h>

#include <math.h>
#include <vector>
#include <string>

using std::vector;
using std::string;

const int SPECT_WIDTH = 64;

ArduiPi_OLED display; // global, for use during signal handling

void cleanup(void)
{
  // Clear and close display
  display.invertDisplay(false);
  display.clearDisplay();
  display.display();
  display.close();
}

void signal_handler(int sig)
{
  switch(sig) {
    case SIGINT:
    case SIGHUP:
    case SIGTERM:
      cleanup();
      break;
  }
}
[... 512 lines cut ...]
    opts.error("could not create cava config file: " +
               string(strerror(errno)));

  // Create a pipe to a cava subprocess
  string cava_cmd = "cava -p " + config_file_name;
  FILE *from_cava = popen(cava_cmd.c_str(), "r");
  if (from_cava == NULL)
    opts.error("could not start cava program: " +
               string(strerror(errno)));

  // Create a file stream to read cava's raw output from
  FILE *fifo_file = fopen(fifo_path_cava_out.c_str(), "rb");
  if(fifo_file == NULL)
    opts.error("could not open cava output FIFO for reading");

  init_signals();
  atexit(cleanup);
  int loop_ret = start_idle_loop(display, fifo_file, opts);

  if(loop_ret != 0)
    exit(EXIT_FAILURE);

  return 0;
}
`

antiprism, 2020-12-01:
Hi Mase

Thanks for the changes. I don't think it needs an option, the spectrum display could just be disabled when the player isn't playing. Could you check the following code works for you

```
int num_bars_read = 0;
    if(select(FD_SETSIZE, &set, NULL, NULL, &timeout) > 0)
      num_bars_read = fread(&disp_info.spect.heights[0], sizeof(unsigned char),
          disp_info.spect.heights.size(), fifo_file);

    if(num_bars_read == 0 || disp_info.status.get_state() != MPD_STATE_PLAY) {
      std::fill(disp_info.spect.heights.begin(), disp_info.spect.heights.end(),
          0);
      usleep(0.1 * 1000000);  // 0.1 sec delay, don't idle too fast if no need
    }
```

Could you also test the following code please. It is to stop the spectrum lagging behing the audio, and also that the spectrum is sometimes "shift-rotated" so the bass bars are in the middle of the spectrum. It should also help clear the microphone data when you start playing audio, if that is an issue.
```
    int num_bars_read = 0;
    if(select(FD_SETSIZE, &set, NULL, NULL, &timeout) > 0) {
      do {
        num_bars_read = fread(&disp_info.spect.heights[0], sizeof(unsigned char),
          disp_info.spect.heights.size(), fifo_file);

        FD_ZERO(&set);
        FD_SET(fifo_fd, &set);
        timeout.tv_sec = 0;
        timeout.tv_usec = 0;
      } while (select(FD_SETSIZE, &set, NULL, NULL, &timeout) > 0);
    }
    
    if(num_bars_read == 0 || disp_info.status.get_state() != MPD_STATE_PLAY) {
      std::fill(disp_info.spect.heights.begin(), disp_info.spect.heights.end(),
          0);
      usleep(0.1 * 1000000);  // 0.1 sec delay, don't idle too fast if no need
    }
```
Adrian.

supercrab, 2020-12-01:
Yes, you're right - having a parameter for it seems silly now.  It definitely makes more sense to just disable the equaliser when the music is not playing.

Just tested and it all works fine :)

Mase

antiprism, 2020-12-01:
Great. I'll add that in!

Adrian.

closed 2020-12-01 by antiprism