antiprism/mpd_oled #50: Is it possible to use 12864 display controlled by SPI ST7567 (same as UC1701)

phenolog, 2020-12-18:
Hello.
There is an interesting display module  FYSETC Mini12864 Panel, controlled by SPI ST7567 (same as UC1701).
Is it compatible with the mpd_oled ?

antiprism, 2020-12-18:
Hi

I don't believe the controller is compatible with any of the controllers supported by mpd_oled, so I would not expect it to work with mpd_oled.

Adrian.

supercrab, 2020-12-18:
Hi

Your screen is supported by the U8G2 library and Adrian did say he planned to use a RPi port of the library...  ***sneaks off***

Mase

antiprism, 2021-07-22:
Hi phenolog

I have made a development version of mpd_oled available, which includes specific support for your display

   https://github.com/antiprism/mpd_oled_dev

Adrian.

antiprism, 2022-11-05:
Closing. Use the mpd_oled_dev repository for further questions.

closed 2022-11-05 by antiprism