antiprism/mpd_oled #51: Radio station name

mikeo999, 2021-01-11:
Is it possible to show the web station radio name on the display.
Using a 4 pin I2C SSH1106 display.

antiprism, 2021-01-11:
Hi

If the web radio station name is available to the player then it should be displayed in mpd_oled when the station is playing.

Adrian.

mikeo999, 2021-01-13:
Thanks that is clear. Is it maybe possible to show the name you gave to a station (i use Moode) . I add some station manually and some have no webstation name. Would be great if you can then use the name you gave the station to display.

antiprism, 2021-01-13:
Hi

On Moode, mpd_oled gets the radio station name from the details provided by Moode in the currentsong.txt file. To check what this contains, start playing your radio station and log into Moode and print the contents of this file to the screen
```
cat /var/local/www/currentsong.txt
```
If this file contains the name of the radio station that you would like to be displayed, copy and paste the contents here and I will see if I can incorporate it into mpd_oled (also, say what version of Moode you are using).

If this file does not contain the name of the radio station, then you could request it as a feature on the Moode forum.

Adrian.

mikeo999, 2021-01-13:
Thanks Adrian for the help.

The file have this info:

`file=http://stream.radio10.nl/radio10
artist=Radio station
album=Radio 10
title=Bruno Mars - Marry You
coverurl=imagesw/radio-logos/Radio 10.jpg
track=
date=
composer=
encoded=VBR
bitrate=128 kbps
outrate=16 bit, 44.1 kHz, Stereo, 1.411 Mbps
volume=100
mute=0
state=play
`

Where Album is the Station name.
I use the latest version moode-r701 version
I rather make my own stream so the stations name look normal instead of lots of extra info
When you use a bunch of streams it gets messy and now i can show it how i like it and find the right one quick.

antiprism, 2021-01-13:
Thanks for posting your currentsong.txt file. I have had a look at what is going on and the issue is that mpd_oled is getting the radio name directly from MPD, which does not provide a name in this case, rather than from Moode, which does provide a name. The correct behaviour on Moode should therefore be for mpd_oled to get the radio station name from from currentsong.txt.

I will fix this and post again when an update is available.

Adrian.

antiprism, 2021-01-14:
Hi

I have just pushed an update that should fix this.

The update required changing the logic of obtaining values generally on Moode, so please been on the lookout for errors when displaying any kind of value and when using any kind of source or renderer.

Adrian.

mikeo999, 2021-01-14:
Hi,

Just did a Git pull but unfortunately it is not working. With some stations it is working fine but the one i show above it is getting worse.

When i start the stream (Radio 10) nothing is show now, no station name and no track info. After a while the track number come back but station name still not showed.

Other stations are random working. Some working direct and others start with same problem (no station and track info) and later about 10 seconds later track info comes back.

Strange things is that bitrate info is also change all the time now for every station. Like 92 - 86 - 82 - 136 etc.

`file=http://stream.radio10.nl/radio10
artist=Radio station
album=Radio 10
title=Simply Red - Something Got Me Started
coverurl=imagesw/radio-logos/Radio 10.jpg
track=
date=
composer=
encoded=VBR
bitrate=128 kbps
outrate=16 bit, 44.1 kHz, Stereo, 1.411 Mbps
volume=99
mute=0
state=play
`

antiprism, 2021-01-14:
Hi

I am running Moode 6.4, and added in the Radio 10 station for testing, and it was working correctly. I'll check on a copy of Moode 7 in case the metadata reporting has changed, although my currentsong.txt looks like yours
```
file=http://stream.radio10.nl/radio10
artist=Radio station
album=Radio 10
title=Bryan Adams - Please Forgive Me
coverurl=images/radio-logos/Radio 10.jpg
track=
date=
composer=
encoded=VBR
bitrate=128 kbps
outrate=24 bit, 44.1 kHz, Stereo, 2.117 mbps
volume=20
mute=0
state=play
```

The delay is probably the delay of Moode updating currentsong.txt. You could check the contents of currentsong.txt repeatedly when changing a radio station, and see if the OLED update is more than a second after the currentson.txt update.

The currentsong.txt for Radio 10 has "encoded=VBR", which means a variable bit rate. mpd_oled gets the bitrate from MPD, which reports whatever the current bitrate is (it should also have been like this before the update).

Adrian.

mikeo999, 2021-01-14:
Hi,

Only the name of the station of Radio 10 is still missing now.
Track info is clear now, thanks for pointing that out.

antiprism, 2021-01-14:
I just tested on Release: 7.0.0 2020-MM-DD, and the "Radio 10" name is showing. I'll update to 7.0.1 and also test that.

mikeo999, 2021-01-14:
That's strange that is working. So then i have to look at my installation. But is nothing more then the image of moode-r701-iso with mpd_oled.

antiprism, 2021-01-14:
Hi

I have installed 7.0.1 and mpd_oled, and added the Radio 10 station, and the OLED is correctly displaying "Radio 10".

The problem may be that you are running an old mpd_oled binary. I recommend making a new clone of mpd_oled, and when you have built it ensure that you run the test command starting with './', e.g.  './mpd_oled -o6', and check if the radio station name is displayed. (If you need to stop an installed version of mpd_oled, run `sudo systemctl stop mpd_oled`).

Adrian.

mikeo999, 2021-01-15:
Hi,

That was it. Installed Moode and Oled fresh and now it is working fine.
Thanks for the very fast support and solution.

Michel

closed 2021-01-15 by mikeo999