antiprism/mpd_oled #52: Integration to Volumio as plugin

misko903, 2021-01-24:
Hello,
I am using this plugin since I my first Raspberry - 3B+ with volumio on it. I have installed display at the beginning too! I love your work.
The only problem is, I must reinstall it everytime after Volumio get update, as Volumio is checking system integrity.
Is it possible to make your plugin an official Volumio plugin to install it simply from repository?

Thank you,
Michal, Slovakia

antiprism, 2021-01-25:
Hi Michal

I'll give a slightly longer answer, to explain the situation and suggest possible workarounds.

mpd_oled needs a copy of the playing audio to show the spectrum. The mpd_oled instructions change /volumio/app/plugins/music_service/mpd/mpd.conf.tmpl, which is a Volumio system file, and I imagine that it is just this change that is causing the system integrity check to fail (I don't think it will care about changes in the home directory or new binaries installed, but I haven't checked). In which case, if you keep a copy of the original mpd.oled.tmpl file, and copy it back before an upgrade, that might be enough.

Looking at the Volumio docs https://volumio.github.io/docs/API/Command_Line_Client.html you may be able to run the following command to remove the custom change
```
updater restorevolumio
```

Similarly, if you use the ALSA method to make a copy of the audio ( https://github.com/antiprism/mpd_oled/issues/4 ), rather than changing the Volumio mpd.conf.tmpl file, there should be no sytem changes  and the upgrade should hopefully work.

I have seen messages on the Volumio forum that suggest it will be easier to make a reliable copy of the audio in the future. I'll review the feature it it becomes available as it should also avoid the integrity check issue.

Regarding creating a plugin, this isn't something I would do. However, I am currently reworking the mpd_oled build so that it is easy to make a binariy that will work on all Pis and supported Player OSs. This will make installation easier, but also makes the job of creating a plugin easier if soeone wishes to do this.

I think it would be useful to keep this issue open. Please report anything that helps with the upgrade issue and I can add it to the installation instructions.

Adrian.

supercrab, 2021-01-25:
I wrote a volumio plugin a couple of years ago and have thought about writing one for `mpd_oled`.  It's on my list of things to do though and I think it would be useful for configuring `mpd_oled` once it has been built by the user.

A volumio plugin is nothing more than a javascript file that can respond to various volumio events.  They save their config in a corresponding JSON file and can run OS commands (I'm sure I saw one that did).  So in theory, if the plugin in theory should be able to adjust the `ExecStart` line in the `mpd_oled.service` file and run `install.sh` to install the changes  (assuming the plugin process has rights to access the folder and execute the script)

Will report back when/if I get a chance to experiment :)

antiprism, 2021-01-25:
Hi Mase

Sounds good, although it might be some work to support all the mpd_oled options in a UI!

Adrian.

supercrab, 2021-01-25:
@antiprism oh, the UI front end of a plugin is defined by a JSON file, so if you know all the available options, it's straight forward enough to do

[https://github.com/volumio/volumio-plugins/blob/master/plugins/system_controller/gpio_control/UIConfig.json](example UI config from my plugin)

misko903, 2021-01-25:
Hi Adrian, hi all,
thanks for longer answer, you explained a lot. Next time I will try to use the ALSA method or even I will try to use `updater restorevolumio` . There is an option in Volumio, where can I restore the system, but It will delete all the custom settings as connected drives, favourite radios... Practically it is clearly installed system. Then I am able to update and the procedures of setting up begins.

One option, instead of 
`sudo nano /boot/config.txt` It is possible to use `sudo nano /boot/userconfig.txt` , which will stay unchanged after restoring the system.

So I'll keep it is for now, but I'll try it on next Volumio upgrade.
Thanks
Michal

@supercrab , this would be awesome!

antiprism, 2021-01-25:
Hi Michal

Thanks for the suggestion. You may have some old instructions for installing mpd_oled on Volumio as the current instructions now include commands for editing   /boot/userconfig.txt

Adrian.

antiprism, 2021-01-27:
Hi Mase

I backported some of my development code into the current mpd_oled and then, after several hours of tweaking build options, was finally able to produce a single mpd_oled binary, on a Pi 4 running Raspbian buster, that will work on both Volumio and Moode on a Pi Zero (I haven't checked RuneAudio, see below).

The build takes a bit longer, especially on a Pi Zero, but I imagine most people will choose a binary install if it is available.

I'll complete this development (I haven't updated the documentation yet), and see if I can write a script to produce a full binary installation (that also includes cava) as a deb file. Hopefully a plugin could then just install this deb file during installation.

Adrian.

Note for RuneAudio: the downloadable version is very old and I don't think it works on a Pi Zero, and people seem to run a number of newer custom images shared through the forum, but it is often hard to install a working compiler and so a binary install of mpd_oled would be especially useful there.

supercrab, 2021-01-27:
@antiprism 

Nice work!  I just checked the documentation of [Volumio plugins](https://volumio.github.io/docs/Plugin_System/Writing_A_Plugin.html) and it does support an install bash script, so that should be able to do the installation.

Here's an example of a plugin install script for illustrative purposes.  It looks like it could be used to clone and compile the source, without the need for binaries or it could do pretty much anything it needs to.

```#!/bin/bash

echo "Installing Spop Dependencies"
sudo apt-get update
sudo apt-get -y install libasound2-dev libreadline-dev libsox-dev libsoup2.4-dev libsoup2.4-1 libdbus-glib-1-dev libnotify-dev --no-install-recommends

echo "Installing Spop and libspotify"

DPKG_ARCH=`dpkg --print-architecture`

echo $DPKG_ARCH
cd /tmp
wget http://repo.volumio.org/Packages/Spop/spop-${DPKG_ARCH}.tar.gz
sudo tar xvf /tmp/spop-${DPKG_ARCH}.tar.gz -C /
rm /tmp/spop-${DPKG_ARCH}.tar.gz

sudo chmod 777 /etc/spopd.conf

#required to end the plugin install
echo "plugininstallend"```

antiprism, 2021-01-29:
Hi Michal

I can confirm that Volumio can be updated if the following command is run first (with the understanding that it may also undo other custom changes not related to mpd_oled)
```
volumio updater restorevolumio
```
All I had to do after the restart was reconfigure the audio copy
https://github.com/antiprism/mpd_oled/blob/master/INSTALL_VOLUMIO.md#configure-copy-of-audio

I'll add this to the instructions.

Adrian.

misko903, 2021-01-29:
Thanks for the info! When I was doing the restore through volumio GUI, it
has deleted all my settings incl. All the radio stations, music sources and
favourites.

I'll try it on next update, it will save me a lot of time :)

I am watching also the work around the official plugin. Impressive!!!

antiprism, 2021-02-02:
Hi Mase

I have just pushed a development branch. You can get a copy with
```
git clone -b dev https://github.com/antiprism/mpd_oled.git
```
I will make a binary installation next, but there are installation changes to help with the binary installation that should also make life easier for you with the plugin.

The instructions now install mpd_oled before testing it, and the installation includes some scripts to install and manage the mpd_oled service. You can run `sudo mpd_oled_install.sh` to install the service (it uses a dummy OLED setting, which causes the mpd_oled command to fail, and the service simply does not run).

The mpd_oled options that the service will use are set by running `sudo mpd_oled_edit.sh` followed by the options, and the service file will be updated and the program started. I think this could also work for testing command line options, rather than running the mpd_oled command directly, because as soon as the options are good the configuration has been set and there is no need to do anything else. (It might be an idea for the script to print any mpd_oled error message.)

I will also script the change to mpd.conf.tmpl. I have asked on the Volumio forum if anyine knows a command to trigger the regenereration of mpd.conf, but have not yet received a reply.

Adrian.

Further note for RuneAudio: it looks like it has been forked, so I will see if I can support the fork: https://github.com/rern/rAudio-1/

supercrab, 2021-02-02:
Thanks for the update!  I'll check it out in a bit :)

So far, I'm about 25% done with the plugin.  I had a few blockers to do with the way Volumio plugins are implemented and some things aren't documented but it should be straightforward from now.

 I've managed to get the plugin to start execution of `mpd_oled` which is promising.  I couldn't get it to run `install.sh`  to install the service because it needs `sudo` authority.  This is strange because I'm using the following line in JS to kill `mpd_oled` before starting it again `exec('/usr/bin/sudo /usr/bin/killall mpd_oled')` 

I haven't looked at automating the install of mpd_oled and configuring the copy of the audio yet...

Cheers
Mase

antiprism, 2021-02-04:
Hi Mase

I have managed to get help on the Volumio forum for a script to trigger the regeneration of /etc/mpd.conf

   https://community.volumio.org/t/command-to-regenerate-mpd-conf/44573

That means I can now write a script which will add the audio_output section to /etc/mpd.conf without user intervention.

However, the system integrity check will fail even by running `touch` on a file, and so for the uninstall script I will see if it is possible to restore of just the mpd.conf.tmpl file (rather than all of /volumio).

Adrian.

antiprism, 2021-02-04:
Hi Mase

I have attached a zip file with a couple of scripts for review. The install script adds the audio_output section to /etc/mpd.conf and the uninstall script removes it again. Run `cat /etc/mpd.conf` after running either script to see it is updated.

The uninstall script uses the technique used by "volumio updater restorevolumio" to just restore the mpd.conf.tmpl file. However, rather than rebooting it just remounts the root filesystem, and this seems to synchronise the overlay filesystem. I don't know enough to say if this is a good idea or not, but it does appear to work. It is probably better not to run it on your favorite Volumio setup just yet! I'll ask on the Volumio forum if someone could review the script.

Adrian.

supercrab, 2021-02-04:
Ooh nice!  Looks good to me!

antiprism, 2021-02-04:
Hi Mase

That is good progress with the plugin, and it will be multilingual too!

I haven't made any binary packages yet, but it is easy to build the binaries. If you look at the last two commands in commands.txt
https://github.com/antiprism/mpd_oled/blob/dev/commands.txt
They are commands to run when configuring mpd_oled and cava, and when you run `make install-strip` afterwards they make binaries that run on a Pi Zero and are installed under /home/volumio/inst. This is the install tree that I will be packaging.

Adrian

misko903, 2021-02-04:
If you need Slovak, maybe Czech translation, I can help.

M

antiprism, 2021-02-04:
I could do the Spanish translation.

supercrab, 2021-02-04:
I could ask a friend to do a translation in Welsh :)  I don't have to make it multilingual but it looks neater and it's a more complete solution.  It is making the code more complex and is taking me some extra time to support it.   It's handy to have all the strings contained in the plugin contained in a separate JSON file, so any changes there will also appear in the UI :)

I did read the installation guide for installing in the end - I found it.  I don't know anything about install trees or much about Linux in general.  Fingers crossed I can get it working, the heat is on!

antiprism, 2021-02-04:
Hi Mase

There is a mistake with quoting in the install script, the variable that sets the text to insert should look like
```
mpd_conf_txt="
# add a FIFO to be read by a cava, which is run as a subprocess of mpd_oled
audio_output {
        type            "\""fifo"\""
        name            "\""mpd_oled_FIFO"\""
        path            "\""/tmp/mpd_oled_fifo"\""
        format          "\""44100:16:2"\""
}"
```
Also, mpd_oled cannot be running when the install script is run. I need to review the conditions for the scripts to work and try to
warn or account for them. One of the problems seems to be whether mpd can open/close/create/delete the FIFO when cava also has it open.

Adrian.

supercrab, 2021-02-05:
It should be possible to use `killall mpd_oled` and the same for mpd.  If MPD is killed how do we restart it?

I have some code tidying, finish up multi language support, finish off the UI JSON, little refactor here and there and then I’ll be onto install and uninstall malarkey. I managed to add a button to the UI which runs `i2cdetect` if it’s present (i need to add this to the install) and it tells you what address your oled screen is. I have to account for multiple devices on the bus on either bus.

antiprism, 2021-02-05:
Hi Mase

MPD will be run as a service, so `systemctl stop mpd` should stop it, and then it can be started again with ` systemctl start mpd` (checking that Volumio doesn't also try to restart it, maybe run `mpc stop` first). mpd_oledcould be running as a service or on the command line. It is possible to check if it is running as a service with `systemctl is-active --quiet mpd_oled`, in which case it can be stopped and started again afterwards. And if it is not running as a service then run `killall mpd_oled` and don't start it again.
I'll add it like this to the install and uninstall scripts for the mpd.conf FIFO.

Adrian.

antiprism, 2021-02-06:
Hi Mase

I forgot to mention, the binary package (which I am going to look at now) will install dependencies, which will include the i2c-tools package (although not a strict dependency) which contains i2cdetect.

These are the updated mpd.conf install/uninstall scripts
[mpd_oled_volumio_scripts2.zip](https://github.com/antiprism/mpd_oled/files/5936633/mpd_oled_volumio_scripts2.zip)

Sometimes, after running them, the first click of the play button doesn't work. They could be quietened a bit more. I haven't checked what happens depending if a non-MPD audio source is playing.

Adrian.

supercrab, 2021-02-06:
Ah yes, I forgot about MPD being a service - should be easy to stop and start then.  

I couldn't get the install script to update the `mpd_oled` service parameters to run from volumio (because of permission errors) but I wonder if I can run each command separately to get around this.

Scripts look good, cheers!

antiprism, 2021-02-06:
Hi Mase

I don't know about the permission errors. Normally the commands in a script run with the same user/permissions that you run the script.

I have an i2c SSH1106 attached (-o 6) and none of the other -o settings cause a segfault for me, but if the wrong model is chosen then I imagine the program could write to memory that it shouldn't and cause a segfault.

Adrian.

supercrab, 2021-02-06:
I think it's to do with the `install.sh` needing `sudo` to work. The current user is `volumio`.  In my tests calling `/usr/bin/sudo install.sh` failed but killing processes with `/usr/bin/sudo /usr/bin/killall mped_oled`  works.  Maybe `install.sh` would work if I preceded each line in the script with `/usr/bin/sudo`?   I'll have a look when I finish the plugin...  I'm sure it's gonna be fun & games..

supercrab, 2021-02-06:
I2C scanner finished,  `UIConfig.json`, install/remove scripts to go...

[image]

supercrab, 2021-02-07:
Hi @antiprism 

I've attached the install and uninstall scripts that the plugin will call for your you to review.  I haven't tested it from the plugin yet as I thought I'd get some feedback first.  It removes `cava` and `mpd_oled` during the uninstall, is that what you would expect?

[plugin-install.zip](https://github.com/antiprism/mpd_oled/files/5938769/plugin-install.zip)

Cheers
Mase

misko903, 2021-02-07:
Hi guys, just an idea - I think there is a plugin already using display
integrated on HAT. Maybe they have some knowledge how to do it? The plugin
is called pirateaudio, but I think nanosound have it too... But maybe I am
late :)

Michal

supercrab, 2021-02-07:
@misko903 I had a look  at the pirateaudio hat and it looks really cool!  It has a colour screen and lots of stuff built onto the hat too.  I had a look at the plugin that configures the screen and it wasn't very configurable - it only had 2 options.  The MPD_OLED application that @antiprism is really configurable, can use different OLED screens and now it'll have a Volumio plugin to help set it up and configure it :)

My plugin should be ready soon :)

antiprism, 2021-02-08:
Hi Mase

Good work on the i2c scanner.

The binary deb package I am making for mpd_oled will install the mpd_oled and mpd_oled_cava (cava) binaries, as well as the support scripts (like the ones that change /etc/mpd.conf) and other support files, as well as automatically installing the other package dependencies. It will also be possible to upgrade the package, and to uninstall the package (which will delete everything that was installed).

If you install the programs from source, the same files (the important ones anyway) will be installed, and to the same place as the package (if not by default then this can be configured). mpd_oled probably supports `sudo make uninstall`, and maybe cava too (but cava only installs two files anyway), which will (should) remove everything installed by `make install-strip`.

The package will be downloaded, and then installed by a command like `sudo apt install ./mpd_oled_0.2_01012021_amrfh.deb`
It will be configured by the user by running scripts like the ones that set up the service, or the ones that set up a FIFO.

The package scripts are ready now, and I just have to adapt the existing documentation. I may have a package ready in the next couple of days.

For the installation of the plugin, you could rely on the package manager for installing and removing mpd_oled (you could download or bundle the mpd_oled package) and then call the installed scripts from your scripts rather than using modified versions. This should make the plugin installation and removal easier to write, and also more robust.

Looking at your scripts and https://volumio.github.io/docs/Plugin_System/Writing_A_Plugin.html , it looks like the plugin install.sh is run as root, but maybe as volumio with sudo, so if you create files under /home/volumio it may be only root can delete them, which you may want. If you don't want to keep the directories afterwards you could work in a temprorary directory and delete it afterwards (example in the script). If anyone has looked at mpd_oled before but not actually installed it then the install script will fail to install mpd_oled. The uninstall doesn't uninstall mpd_oled from the system, but it might be better to leave it if a package manager is not involved.

If you need root privileges for settings but are not doing this from the install script and so the script is running as volumio, it is probably possible to make a small helper binary that will run as root and apply the settings: https://www.computerhope.com/jargon/s/setuid.htm . I would add this to the binary package.

Adrian.

antiprism, 2021-02-08:
Hi Mase

I was originally going to implement configurable layouts first, and I have already written the difficult code for this, but it requires lots of other changes before producung a working OLED display. I decided instead to reverse the order and add the layouts last, so I can introduce changes incrementally.

The currently planned steps are:

1. Add the autotools build and make a binary that can work on all players (almost complete)
2. Switch to U8g2 for the OLED contoller and graphics
3. Add a few fixed layout variations (with and without spectrum) and support all common OLEDs
4. Add configurable layouts

U8g2 is monochrome only, but I'll look at whther colour displays can be supported, as I have one available.

Adrian.

supercrab, 2021-02-08:
Yeah, it's a lot of work isn't it?  Essentially a rewrite to support all this stuff and implement U8g2. 

I've managed to get the plugin into a state of ready to test but for some reason the `plugin publish` command isn't working as expected and it's not publishing my changes to github. :(

supercrab, 2021-02-08:
New volumio plugin is ready for testing!

1) Activate SSH on your system : SSH
2) Connect to your device via your console (linux) or Putty (Windows)
3) Download the plugin:
`wget https://github.com/balbuze/volumio-plugins/raw/master/plugins/audio_interface/volparametriceq/volparametriceq.zip`

4) Now create a temporary folder `mkdir ./mop`
5) Unzip the previously downloaded file miniunzip `mpd_oled.zip -d ./mop`
6) Go to this folder `cd ./mop`
7) Install the plugins `volumio plugin install` and wait - It will take several minutes to do the building, then:
8) Remove all temporary files : `cd .. rm -Rf mop`
9) In the webUI, go to installed plugins section and enable the plugin.
10) Go in the plugin settings, change Oled type, hit save!

Plugin install

Won't change mpd_oled folder if it already exists
Won't change cava folder if it already exists

Plugin uninstall

Removes mpd_oled folder
Removes cava folder
Remove mpd conf changes

misko903, 2021-02-08:
So if I want to test it, I will leave everything as it is (display is
working now) and proceed with steps you wrote. Right?

Michal

supercrab, 2021-02-08:
@misko903  yes, follow the install steps :D  CAVA and mpd_oled won't be changed so it should be a pretty quick install. :)

misko903, 2021-02-08:
The link in step 3 is wrong. I will search for the right one.
Also the miniunzip word in step 5 should be styled as code

miso

misko903, 2021-02-08:
sorry for spamming, I see, on GitHub it is already corrected. I red the
email only...
miso

ut 9. 2. 2021 o 0:16 miso <misko0903@gmail.com> napísal(a):

supercrab, 2021-02-08:
Ah yes, I updated the page when I tested the steps.  No worries.

misko903, 2021-02-08:
Excellent, sirs!
on my RPI3B+ works everything ok, as now. Do you want some debug info? How to obtain it? My config is exactly as descridbed on webpage, but I have 20 bars + Reversed display.

supercrab, 2021-02-08:
Awesome!  Just give it a good test of the settings if you can.  You'l have to type `jounalctl -f` and it will list some debug info but if `mpd_oled` is running the log will get filled up quickly.  You might find it easier to disable the plugin, make your changes in settings and then press Save, you'll see what parameters mpd_oled is being called with.

What parameters did you use before with mpd_oled?

misko903, 2021-02-08:
`./mpd_oled -o 6 -b 21 -g 1 -f 20 -R`

supercrab, 2021-02-08:
@misko903 have you changed the settings in the plugin to match?

misko903, 2021-02-08:
I love the settings - especially scrolling, as before the scrolling was too slow for me.

Just one thing about which i am thinking longer - OLED display is degraded after 2.5 years of using. Maybe adding a screen saver option will help to reduce the wear? Should I create a new ticket?

misko903, 2021-02-09:
> @misko903 have you changed the settings in the plugin to match?

Yes, it is working also with my original` ./mpd_oled -o 6 -b 21 -g 1 -f 20 -R` settings

supercrab, 2021-02-09:
There is an invert option and timeout period (in hours) that will invert the screen periodically. Maybe @antiprism could change it from hours to minutes sonic inverts quicker?

I don’t think these cheap little screens are designed to be open for a long time, so perhaps a solution could be to clear the screen if no music has played in a certain while?

antiprism, 2021-02-09:
Hi Mase

Congratulations on getting plugin into an installable state! I'll take a look later today.

I have made a first binary package, and tested it on a copy of Volumio on a Pi Zero with a factory reset. It can be installed like this
```
# install package
sudo apt update
sudo apt install gdebi-core
(cd /tmp && wget http://antiprism.com/misc/mpd-oled_0.2+202102081602+ed17ee2_armhf.deb)
sudo gdebi -n /tmp/mpd-oled_0.2+202102081602+ed17ee2_armhf.deb

# does it work?
sudo mpd_oled_volumio_mpd_conf_install.sh
mpd_oled -o 6
```
Unfortunately, although 'apt' (or 'aptitude') should be able to install the remote package, it doesn't on Volumio. 'apt-get' should be able to install the local package, but doesn't on Volumio. 'gdebi' works, but isn't installed by default and is only for installing local package files. The commands therefore go the roundabout way of installing gdebi, downloading the mpd_oled package and then installing the downloaded mpd_oled package with gdebi.

The install also installs the dependencies. I added i2c-detect to recommended packages, but it did not install, so I will add it to dependencies.

The package is removed with
```
sudo apt remove -y mpd-oled
```
I will finish udating the documentation and then merge the development into the master branch.

Adrian.

antiprism, 2021-02-09:
Hi Michal

I'll add a screen saver.

The scroll is configurable  on the command line with `mpd_oled -s`.

You can see all the options by running `mpd_oled -h`, but it would probably be helpful if I listed them in the documentation, so I'll do that.

Adrian.

antiprism, 2021-02-09:
Hi Mase

The invert option will take a decimal number, e.g. `mpd_oled -I 0.001` inverts the display every 3 seconds or so. I'll document this if it is not clear. I used hours as I didn't think people would want to see the screen changing often, but that was just a guess.

Adrian.

misko903, 2021-02-09:
maybe we can add it to settings on volumio GUI?
which file should be translated? I can start with Slovak, maybe Czech...

supercrab, 2021-02-09:
@antiprism Cool, I didn't realise invert took a decimal, I thought it was just integers.  I'll add that to the user config, does 0.001 = 3.6 seconds?  I found this in the code `(period > 0) ? (fmod(time(0) / 3600.0, 2 * period) > period) : period;` but I wasn't sure what the intention is.  I'll try the latest binaries now..

supercrab, 2021-02-09:
@misko903 I've attached a language file for translation :)  Thanks

[strings_sk.zip](https://github.com/antiprism/mpd_oled/files/5950948/strings_sk.zip)

antiprism, 2021-02-09:
Hi Mase

The period can be - 0: always normal, -1: alway inverted, positive decimal: invert with that period. For fmod(time(0) / 3600.0, 2 * period), time(0) gives the current time in seconds (relative to some date) and then get the remainder when divided by twice the period.

So, if period is 0 or -1 then period is returned, and 0=false and -1=true (non-zero). For a positive period, the remainder is either less than the period=false, or greater than the period=true, o it is normal then inverted each for the length of the period.

Not especially clear! I'll comment it, but also, maybe it doesn't all have to be just one line...

Adrian.

supercrab, 2021-02-09:
I've tested the parameters for it and seems to work fine.  So would a value of 0.001 refresh every 3.6 seconds?

antiprism, 2021-02-09:
Hi Mase

-I 0.001 is a period of 3.6 seconds, but the display only changes on whole numbers of seconds, so the inverting isn't very regular at that frequency.

Adrian.

supercrab, 2021-02-09:
I renamed the script as `install2.sh` but didn't get anything on the spectrum analyser, mpd.conf has the correct changes.

` bash install2.sh
{"time":1612873146004,"response":"stop Success"}Failed to stop mpd.service: Access denied
Warning: Stopping mpd.service, but it can still be activated by:
  mpd.socket
rm: cannot remove '/tmp/mpd_oled_fifo': No such file or directory
`

supercrab, 2021-02-09:
i stopped it, changed the audio settings and it seems to be working now.

supercrab, 2021-02-09:
volumio@volumiotest:~/mop$ sudo bash install2.sh
{"time":1612873970631,"response":"stop Success"}rm: cannot remove '/tmp/mpd_oled_fifo': No such file or directory
volumio@volumiotest:~/mop$ mpd_oled -o 3

antiprism, 2021-02-09:
The error mssage the first time said "Access denied", probably because of not using sudo.

The other eror messages always occur. I will quieten them after a testing period.

If mpd_oled is running as a service then there may be issues with the order that the services run. I would think that MPD should be running before mpd_oled, so the install script might need to enforce that by waiting for MPD to be active.

I am going to rename the three service scripts mpd_oled_install.sh -> mpd_oled_service_install.sh, etc, for clarity.

misko903, 2021-02-09:
> @misko903 I've attached a language file for translation :) If you want to to a Czech one then it's you just need to copy the file and change the filename to `string_cs.json`
> 
> [strings_sk.zip](https://github.com/antiprism/mpd_oled/files/5950948/strings_sk.zip)
> 
> **But I wouldn't do any translation just yet as I think I might have some more changes to come...**

Translated. Here you go!

[strings_sk.zip](https://github.com/antiprism/mpd_oled/files/5951611/strings_sk.zip)

[46 more replies]

closed 2022-11-05 by antiprism