antiprism/mpd_oled #53: Can't install

Zipciok, 2021-01-29:
I have problems with instaling this oled screen https://allegro.pl/oferta/wyswietlacz-oled-0-96-ssd1306-blue-arduino-spi-i2c-7636893024 Can somebody make video how to install this  oled on raspbeery pi (I have rpi 3 b)

antiprism, 2021-01-30:
Hi

Give some information about: your operating system, how far you got through the installation instructions, and what when wrong, Copy and paste any error messages.

Adrian,

Zipciok, 2021-01-30:
Hi I use newest Volumio version 
Error:
volumio@volumio:~$ git clone https://github.com/karlstav/cava
fatal: destination path 'cava' already exists and is not an empty directory.
I try to reinstall volumio but it didn't work

antiprism, 2021-01-30:
The git clone command creates a new 'cava' directory in the current directory and copies the cava code into it. The error message says that that the 'cava' directory already exists, most likely because you have previously run this command.

If you have already installed cava then the following command will succeed (print some help text) and you can move onto the next section fof the instructions
```
cava -h
```
Otherwise, delete the cava directory and run the git clone command again (or run the install instructions in a different directory which doesn't have a 'cava' subdirectory).

Zipciok, 2021-01-30:
Thanks for your fast anserw

[image]

There is evry think okey?

antiprism, 2021-01-30:
The instructions don't say to add ``` dtparam=i2c_arm=on``` to /boot/userconfig.txt. It is probably included in /boot/config.txt, so check there first. 

You have missed a 0 in the other line. It should be
```dtparam=i2c_arm_baudrate=400000```
(but you could also try 800000).

Zipciok, 2021-01-30:
Thanks  for anserw, I think it will be better if i delte Volumio from SD card and instal again becaus I think I make a lot of mistakes and some times a have  issue with me english skiill

Zipciok, 2021-01-30:
I Reinstall Volumio and now i have this error but before everythink was okay. 
[image]
When display should on? My never do it.
[image]
Can you make Video tutorial how to install this?

antiprism, 2021-01-30:
For the first problem, you are running the same command twice. You are in the mpd_oled directory after the first comand.

For the scond problem, your I2C display has not been detected by Volumio. Make sure the OLED is connected correctly and check the I2C configuration section of the mpd_oled installation instructions. If your OLED is not detected mpd_oled cannot work. My output looks like this
```volumio@volumio9:~/proj/mpd_oled_dev$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

Zipciok, 2021-01-30:
I wire up this again corecly  and still not working, i think my gpio pin are dead

Zipciok, 2021-01-30:
Thanks for your help and commitment

antiprism, 2021-01-31:
Thanks for letting me know. I'm sorry it didn't work out.

Adrian.

closed 2021-01-31 by antiprism