antiprism/mpd_oled #54: Configurable Date Format

rgmcleod52, 2021-02-11:
I noticed that on the 'stop' screen the date format is DD-MM-YYYY.  Would it be possible to add an option to make the format MM-DD-YYYY for those of us used to that date format?

Thanks,

Rick

antiprism, 2021-02-12:
Hi Rick

Thanks for the suggestion. It is a good idea to support this. I'll add it in.

Adrian.

rgmcleod52, 2021-02-12:
Adrian,

Thanks.  And thank you for all the work you've put into the mpd_oled
project.

I followed your manual installation of the mpd_oled code and got it working
yesterday.  Then today I saw supercrab's plugin work.  I've tried that now
too and it works fine.

Between you and supercrab we've got a nice plugin for Volumio.

Rick

supercrab, 2021-02-12:
Much appreciated @rgmcleod52!  I'm glad everything's working for you!

antiprism, 2021-02-15:
Hi Rick

I have just added in a new option -d, which sets the date format to MM-DD-YYYY.

Adrian.

supercrab, 2021-02-15:
I've updated my plugin to support this @rgmcleod52.

You'll need to uninstall the old plugin and re-install it, so that it can rebuild CAVA.

rgmcleod52, 2021-02-15:
Adrian and supercrab,

Thank you!  The new update works great!

Notes:

1.  I initially tried following the instructions on how to 'refresh' (i.e.
'volumio plugin refresh') the plugin, but then realized that by
uninstalling the plugin, the code that was there wasn't anymore.  So I
found supercrab's initial instructions on installing the plugin (using
'volumio plugin install') and that worked just fine.  I should have
realized that since Adrian had to change his code, it meant an uninstall
and reinstall instead of just a refresh.

2.  I noticed that on supercrab's github page that you say " *When you run
the volumio plugin install command and it tells you the plugin has been
successfully installed, you will need to press Ctrl + C to finish the
process.*"  The last block of comments in the install process starts
with "*Progress:
100 Status :MPD OLED Successfully Installed, Do you want to enable the
plugin now?*"  and ends with "*Done<br>Finalizing
installation<br>Finalizing installation*".  My initial installation was
done via full screen using an HDMI monitor and keyboard, so I could see the
first part of the last block of comments.  But when doing this recent
install, I did it over Putty so my screen size was much smaller, so the
initial part of that last block is off the screen.  Admittedly, I could
have scrolled back (which I did) to see the "Progress: 100" comment.  But
it might be worthwhile to modify your note to look for " *Done<br>Finalizing
installation<br>Finalizing installation*" since that's what the user will
see most easily.  Not a big deal, but it might reduce the number of people
who get confused.

3.  After installing, activating and configuring the new version of the
plugin - with great success! - I did notice that Volumio reports the
version of the plugin as "0.0.1" which is what it was before I uninstalled
and re-installed.  Not sure if this version number comes from Adrian, or
from supercrab, but it would be nice to update the version shown to reflect
even these early days changes.

Guys, I honestly didn't expect to see this update this quickly.  Thank you
for the quick response and teamwork!  Keep up the good work!

Yesterday I built up a NAS using a RPi so that my two Volumio units - one
with the OLED and one with a touchscreen - can feed off of the same set of
music sources.  Now that the OLED is up and running, I'll be designing a
box for the RPi, OLED and some control buttons (via another Volumio plugin)
that I'll 3D print.

Rick

supercrab, 2021-02-15:
Hi @rgmcleod52

1. Yes, mpd_oled needs to be downloaded and rebuilt.  It's the onyl way :(  I did say to uninstall and re-install the plugin though :p

2. The volumio plugin program has a bug.  It should exit when a plugin has installed and let the user get on with it.  Unfortunaltey, only the volumio team can fix this one.  In a couple of days, I'll submit the plugin to the volumio team so that it can be installed using the UI, although I have no idea when it will be included.

3. Good point!  I need to update the version number - I keep forgetting.  I shall do this now actually.

Thanks for taking the time to test and letting me know all the issues you've found :)  It's much appreciated.  Please send pics of your work so I can see :)  Either on volumio community forums or on the Facebook groups.  I'd like to see what people make.

Cheers
Mase (supercrab)

rgmcleod52, 2021-02-15:
Just posted a topic on the Community DIY section that documents my
touchscreen build.

I did mention both of you in regards to work on this new project.  Thanks
so much for being so responsive!

Rick

antiprism, 2021-02-17:
Hi Rick

Thanks for letting me know that the new option is working correctly for you. I'll close the issue!

Adrian.

closed 2021-02-17 by antiprism