antiprism/mpd_oled #55: Volumio buster build parameters need tweaking

supercrab, 2021-02-14:
From my post in the Volumio community there appears to be an issue with compiling on buster, found by user `Darmur`

https://community.volumio.org/t/plugin-mpd-oled-installation-configuration-plugin/44823/30

"tried on volumio buster, it does not work, the issue is related to this command in your build script"

`PLAYER=VOLUMIO make`

on buster-based systems, it should be changed to

`PLAYER=MPD LDLIBS="-li2c" make`

supercrab, 2021-02-14:
D'oh, it's mentioned in the read me!

I think I can do `if cat /etc/os-release | grep buster; then...`

closed 2021-02-14 by supercrab

antiprism, 2021-02-14:
Hi Mase

The command (if it works) should be
```
PLAYER=VOLUMIO LDLIBS="-li2c" make
```

The underlying issue is fixed by the new build system, which detects the player and the libraries required when building.

Fortunately, the binary package, which I build on Jessie without -li2c, also works Buster.

Adrian.

supercrab, 2021-02-14:
Hi

I've updated the `install.sh` to fix this in the plugin :D

What is the new build system?

Mase

antiprism, 2021-02-14:
Hi Mase

The new development uses the Autotools for the build, and will require some changes to your script, and the build takes a long time to complete for people on a Pi Zero. Here it is if you want to take a look

   https://github.com/antiprism/mpd_oled/tree/dev2

You might want to consider using your own copy of the mpd_oled code for the plugin (e.g. fork the mpd_oled repository on Github) then the plugin will not be disrupted by any changes I make.

Adrian.

supercrab, 2021-02-15:
@antiprism OK, I'll make a fork for the plugin, keep things separate.

Cheers