antiprism/mpd_oled #59: Spectrum analyser shows low levels and regularly goes blank

supercrab, 2021-02-25:
Hi

I've compiled the latest version and the spectrum analyser goes blank every 2 or 3 seconds.  The spectrum has low levels and will then display no audio sporadically.  I recompiled CAVA using the old method but the issue persists.

I encountered this when I tested the prebuilt version a few weeks ago on a different pi.  The hardware is fine is good.

Cheers
Mase

antiprism, 2021-02-25:
Hi Mase

I can see this, I thought it was a PI Zero thing, maxing out the CPU. It is quite variable for me, sometimes it happens evey second, other times there can be 20 seconds or more. Also, the scrolling text animation is not smooth, and appears to stutter when the spectrum flashes. I am seeing this on Moode (same deb package as Volumio), and when reading from a FIFO and from an ALSA stream. I don't recall any changes that should especially affect this, but there were a lot of changes so there may be something incorrect. I'll do some investigating!

The spectrum levels are fine for me, but sometimes higher and sometimes lower. I think Cava adapts the levels from time to time.

Adrian.

antiprism, 2021-02-25:
Hi Mase

I had a copy of the old mpd_oled code around and it showed the same behaviour. I then checked out a copy of Cava from a few months ago and installed that, and the flashing has almost completely stopped (I see the occaisional flicker)
```
git clone https://github.com/karlstav/cava.git
cd cava
git checkout -b old e9abc9d3d89604d77e43f7af69acf9ab6f3c943d
```
I'll try to find the Cava commit where this changes.

The text scrolling still isn't completely smooth, so maybe it never was.

Adrian.

supercrab, 2021-02-25:
I've reinstalled the older version of `mpd_oled` and the regular CAVA build and it's fine again!  It's a real head scratcher.  When everything is OK the visualiser remains the same overall level, as my track is a constant volume.  

It's almost as if the data from the CAVA pipe is nonsense.  I'm not even sure if the visualiser is activated by the music.  I will do some digging though!

antiprism, 2021-02-26:
Hi Mase

I have tracked down the Cava commit where the flashing starts

   https://github.com/karlstav/cava/commit/3006603628af84698dd1d87f22af4633446f4cbb

Fortunately the changes are small. I'll ask Karl about it.

Adrian.

supercrab, 2021-02-26:
Hi Adrian

Thanks a lot for that.

Cheers!
Ma

antiprism, 2021-02-27:
Hi Mase

Do you still see the flashing if you disable the mpd_oled scrolling (-s 0)?

Adrian.

antiprism, 2021-02-27:
I have still seen some flashing with -s 0, but also some radio stations seem worse than others, and sometimes there is no apparent flashing for longer periods. There also seems to be an issue with the fluidity of the bar movement (mini freezes) that may be related. I think the best way forward is a review of the mpd_oled timing code.

Adrian.

supercrab, 2021-02-27:
Hi Adrian,

I don’t have that version installed on my pi any more. 😪

Did you do many changes to the code base in the latest version of `mpd_oled`?

antiprism, 2021-02-27:
Hi Mase

I'll check, but the code changes weren't meant to change the program operation, and probably don't change the timing directly. I am also seeing the same issue with the original code and the current cava, although the degree of flashing I see in all cases seems a bit unpredictable.

The code changes incorporated some alternative code for fetching files over http and decoding json, and the code for different players is now conditional during execution, rather than condtionally included when being built (so the same binary can work on different players). It is possible that some operations take a different amount of time, which might affect the overall timing, perhaps making an existing issue worse.

Adrian.

antiprism, 2021-02-27:
Hi Mase

I have had a look at the timing code, and have found a change that is stopping the flashing for me. In main.cpp, could you try modifying the corresponding line to look like this
```
update_sec * 1100000; // slightly longer, but still less than framerate
```
Adrian.

antiprism, 2021-02-27:
Just to note, different radio stream require considerably different amounts of processing. The Radio 1 128k stream (for outside the UK) leaves the Pi Zero with as little as 40% idle, but the Ancient FM - Mediaeval and Renaissance Music 128k stream has a minimum of 55% idle, which is the difference between maxing out the Pi Zero or not when running mpd_oled.

supercrab, 2021-02-27:
When you say “flashing” do you mean the spectrum analyser?

I’ve found using pi 0 to be only good for volumio without any oled. I use a pi3 with the screen 😀

antiprism, 2021-02-27:
Yes, the "flash" when the spectrum isn't displayed.

antiprism, 2021-02-28:
Hi Mase

My previous message doesn't read very well. In case it wasn't clear, there is a statement in main.cpp
```
 const long select_usec =
      update_sec * 1001000; // slightly longer, but still less than framerate
```
If you change it to
```
 const long select_usec =
      update_sec * 1100000; // slightly longer, but still less than framerate
```
Then it appears to stop the spectrum from blanking momentarily (on my Pi Zero).

Adrian.

antiprism, 2021-03-06:
I have pushed this change as I am seeing no blanking with it.

supercrab, 2021-03-08:
Thanks!  I’ll will let you know if it work when I get a setup.

supercrab, 2022-03-01:
Pretty sure this is sorted :)

closed 2022-03-01 by supercrab