antiprism/mpd_oled #60: How to show only Spectrum

alain93, 2021-02-26:
Hi Adrian,
I installed mpd_oled on Moode and their I modified main.cpp and display.cpp to show only the spectrum and bits when playing. Now on Volumio, I installed it from source, it worked well.
Then I went in Home/volumio/src to edit main and dispaly files like i did in Moode, after saving, the screen it did not change, still showing the full set, spectrum, time and all icons.
After modif, I rebooted the system, not sure if there is anything else to do like in Moode.
Or is there another place were the files are?

antiprism, 2021-02-26:
Hi Alain

After changing the source files you need to run `make` and `sudo make install-strip` again.

Adrian.

alain93, 2021-02-26:
Perfecto,
 thanks Adrian

closed 2021-02-26 by alain93