antiprism/mpd_oled #64: Use of Timezone and oled display

rgmcleod52, 2021-03-19:
I looked through the other issues and didn't see this one.

When I initially configured mpd_oled on my RPi, I didn't set the timezone.  So when I manually changed the time via the 'date' command, the display changed accordingly.

But I had to rebuild the OS and reinstall everything including mpd_oled.  But this time I set the timezone to my local timezone (America/Los Angeles), and now while typing in 'date' on the command line gives me the correct time and date, the time and date shown on the oled display always shows UTC time.

To prove it (to myself before firing off this issue) I backed out the timezone setting in the .profile, rebooted and then manually changed the time using 'date'.  Now the display shows the correct time, but the system thinks that this is UTC time.

Maybe I missed the right way to set my timezone and yet have the system know what UTC time actually is.  If so, please let me know.

If not, could you make it possible for the time/date to reflect the local time instead of UTC always?  This isn't an urgent fix, but some people might find it useful.

Not sure if this will require a change to the plugin code that supercrab has created.  I'll let you two hash that out.

Thanks!

Rick

antiprism, 2021-03-19:
Hi Rick

The method for setting the timezone in the mpd_oled installation instructions for Volumio is
```
sudo dpkg-reconfigure tzdata
```
Does that get mpd_oled showing your local time?

Adrian.

rgmcleod52, 2021-03-19:
Yes, that fixes the issue!  I must have missed that command when I was reinstalling the code.

Glad that it was easy.  Sorry for the trouble.

Rick

closed 2021-03-19 by rgmcleod52