antiprism/mpd_oled #65: Support an audio copy using ALSA on Moode (>= 7.1)

antiprism, 2021-03-19:
There will be no more patches for Moode starting from Moode 7.1, and so the copy will need to be made using an ALSA configuration.

The following instructions describe how to set up the ALSA configuration manually. When I have some feedback on how this is working out for other people I will provide scripts to enable of disable the audio copy, so no files will have to be edited by hand.

Here is an `/etc/asound.conf` file for Moode which intercepts the ALSA configuration for the graphic equalisers on Moode 7. It should be used as-is (if no other program is using the loopback device). If the file already exists you should review its contents, and if there are no lines starting "pcm.!eqfa12p" or "pcm.!alsaequal" then you can append the following lines at the end of the `/etc/asound.conf` file (or if either of these lines already exist then post your `/etc/asound.conf` file here)

```
pcm.!eqfa12p {
    type copy
    slave.pcm "eqfa12p_and_copy"
}

pcm.eqfa12p_and_copy {
    type plug
    slave.pcm {
        type multi
        slaves {
            a { channels 2 pcm "plug_eqfa12p" }       # the original output
            b { channels 2 pcm "plughw:Loopback,0" }  # the loopback driver
        }
        bindings {
            0 { slave a channel 0 }
            1 { slave a channel 1 }
            2 { slave b channel 0 }
            3 { slave b channel 1 }
        }
    }
    ttable [
        [ 1 0 1 0 ]   # left  -> a.left,  b.left
        [ 0 1 0 1 ]   # right -> a.right, b.right
    ]
}
pcm.!alsaequal {
    type copy
    slave.pcm "alsaequal_and_copy"
}

pcm.alsaequal_and_copy {
    type plug
    slave.pcm {
        type multi
        slaves {
            a { channels 2 pcm "plug_alsaequal" }     # the original output
            b { channels 2 pcm "plughw:Loopback,0" }  # the loopback driver
        }
        bindings {
            0 { slave a channel 0 }
            1 { slave a channel 1 }
            2 { slave b channel 0 }
            3 { slave b channel 1 }
        }
    }
    ttable [
        [ 1 0 1 0 ]   # left  -> a.left,  b.left
        [ 0 1 0 1 ]   # right -> a.right, b.right
    ]
}
```
In `/etc/modules` I have the following line to load the looppback device
```
snd-aloop
```
The Loopback device loads as first sound card for me at boot, which would mean no playback ausio, and so I force it to be not first by creating the file `/etc/modprobe.d/alsa-base.conf` and adding the line
```
options snd-aloop index=-2
```
Reboot and **enable either graphic equaliser**, then run mpd_oled with a command like
```
mpd_oled -o6 -c alsa,hw:Loopback,1
```

Please post some feedback below if you use this configuration.

Adrian.

merlin699, 2021-03-20:
It nearly works. I followed your steps, but on running mpd_oled command the display went berserk. It flips and jumps and is totally scattered.
In some fractions of seconds I can see that the spectrum visualizer works though.
Took a while to realize that I need to sudo the mpd_oled, in my case:
`sudo mpd_oled -o1 -c alsa,hw:Loopback,1`
I tried all display parameters from my previous version, which were working perfectly with my SPI display:
`sudo mpd_oled -o 1 -b 15 -g 1 -f 25 -c alsa,hw:Loopback,1`
wich caused the same behaviour.

antiprism, 2021-03-20:
Hi merlin699

Thanks for the feedback. I received a report of "flickering" for an SSD1309 I2C, which also uses the SSD1306 driver, I don't know if this is the same issue

   http://moodeaudio.org/forum/showthread.php?tid=155&page=42

Could I just check, is there any issue with the display outside of the spectrum area? If there is, do you also have a problem with
the following command (which ensures that no copy of the audio is available for the spectrum analyser)

   sudo mpd_oled -o 1 -b 15 -g 1 -f 25 -c fifo,dummy
 
Adrian,

merlin699, 2021-03-20:
Hi Adrian,
thanks for your suggestion. But I did a clean install from scratch now.
Once again following exactly your instructions and **it works perfectly now :)**
Maybe there was a hicup somewhere in the update process?!? Anyway, I have another machine with the same display, I'll check if there's the same behaviour.
Thanks again for help and this great piece of software.

supercrab, 2021-03-20:
If the display goes beserk it could be if two versions of `mpd_oled` are running. I can’t remember how I managed to do this but I have seen it. A quick `killall mpd_oled` will fix it.

merlin699, 2021-03-20:
@supercrab looks like you're right. On my second install there were no such troubles after `killall mpd_oled` running fine now. Thx for your help.
[image]

antiprism, 2021-03-20:
Great, thanks for letting me know it is working now, Good suggestion from Mase.

Adrian.

alain93, 2021-03-20:
Hi Adrian,
the install is working great, bar's height seems to be related to Moode's volume. 
Is "sudo mpd_oled_service_install" still a good script to make the display stick?

antiprism, 2021-03-20:
Hi Alain

I have just replied on the Moode forum how to install mpd_oled s a service.

Is the bar height not adapting for you over time.?It does for me. The bar range can start very low, but gets heigher over time.

Adrian.

alain93, 2021-03-20:
ok I will check that.

antiprism, 2021-03-21:
I have prepared scripts for installing and uninstalling the ALSA configuration that makes the audio copy, if anyone would like to try or review them.

I have tried to make them safe against changing existing configurations, but this means extra warnings for existing or previous complete or partial compatible configurations. Any feedback on the scripts' behaviour or messages would be appreciated (if only to say that there was no issue using them).

Here is the first version of the scripts
[mpd_oled_moode_audio_copy_scripts.zip](https://github.com/antiprism/mpd_oled/files/6177922/mpd_oled_moode_audio_copy_scripts.zip)

Install command
```
sudo bash mpd_oled_moode_audio_copy_install
```
Uninstall command
```
sudo bash mpd_oled_moode_audio_copy_uninstall
```
Adrian.

supercrab, 2021-03-21:
No worries 😃 
Nice work on the scripts!

svratonzo, 2021-03-22:
Do you think that enabling graphic eq (with flat curve or so on) will it affects sound i will hear or sound quality?

antiprism, 2021-03-22:
Hi svratonzo

I don't know. Perhaps someone on the Moode forum might know.

If it was an issue, you could probably modify the ALSA configuration and change the following line
```
a { channels 2 pcm "plug_alsaequal" }     # the original output
```
to write to your hardware devide rather than pug_alsaequall, e.g.
```
a { channels 2 pcm "hw:0,0" }     # the original output
```
Which would then bypass the graphic equalizer (even though it was set in the UI).

Adrian.

antiprism, 2021-03-22:
Hi svratonzo

Modify `/etc/asound.conf`

Adrian.

svratonzo, 2021-03-22:
> Hi svratonzo
> 
> Modify `/etc/asound.conf`
> 
> Adrian.

it must be as this?

pcm.eqfa12p_and_copy {
    type plug
    slave.pcm {
        type multi
        slaves {
            a { channels 2 pcm "plug_eqfa12p" }       # the original output
            b { channels 2 pcm "plughw:Loopback,0" }  # the loopback driver
        }
        bindings {
            0 { slave a channel 0 }
            1 { slave a channel 1 }
            2 { slave b channel 0 }
            3 { slave b channel 1 }
        }
    }
    ttable [
        [ 1 0 1 0 ]   # left  -> a.left,  b.left
        [ 0 1 0 1 ]   # right -> a.right, b.right
    ]
}

pcm.!alsaequal {
    type copy
    slave.pcm "alsaequal_and_copy"
}

pcm.alsaequal_and_copy {
    type plug
    slave.pcm {
        type multi
        slaves {
            a { channels 2 pcm "hw:0,0" }      # the original output
            b { channels 2 pcm "plughw:Loopback,0" }  # the loopback driver
        }
        bindings {
            0 { slave a channel 0 }
            1 { slave a channel 1 }
            2 { slave b channel 0 }
            3 { slave b channel 1 }
        }
    }
    ttable [
        [ 1 0 1 0 ]   # left  -> a.left,  b.left
        [ 0 1 0 1 ]   # right -> a.right, b.right
    ]
}
# MPD_OLED_END_TAG: DO NOT CHANGE THIS LINE

antiprism, 2021-03-22:
Hi svratonzo

If you only changed the one line and your playback device is hw:0,0 then that may work.

Adrian.

svratonzo, 2021-03-22:
> Hi svratonzo
> 
> If you only changed the one line and your playback device is hw:0,0 then that may work.
> 
> Adrian.

Ho can i verify which playback device is in use?

antiprism, 2021-03-22:
Hi svratonzo

I don't know. Have you tried searching the Moode forums?

Adrian.

svratonzo, 2021-03-23:
> Hi svratonzo
> 
> I don't know. Have you tried searching the Moode forums?
> 
> Adrian.

with "aplay -l" command i had:

card 3: C20 [CA CXA81 2.0], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

so i put in asond.conf this:

a { channels 2 pcm "hw:3,0" }     # the original output

Moode plays correctly so it seems to work. I dont now if will bypass eq, but works!
Thank you so much

antiprism, 2021-03-23:
Hi svratonzo

Great. Thanks for letting me know that it worked out. (If you change the equalizer settings to someting extreme and the audio does not change then you can confirm that the equalizer is not being used).

Adrian.

antiprism, 2021-03-30:
I have added the audio copy install scripts to mpd_oled, and updated the install instructions for Moode 7.

antiprism, 2021-05-02:
Moode 7.2 includes an option in the UI to configure an ALSA loopback copy of the audio, so where possible it is best to use mpd_oled with Moode 7.2 or greater.

I'll close the issue, as it is solved for Moode 7.0 and 7.1, and no longer needs a solution for Moode 7.2+, but I will keep a link to the issue in the instructions for now and so there may be some additional posts.

closed 2021-05-02 by antiprism