antiprism/mpd_oled #66: Volumio 2.878, 2.882 - installation from source fail

ArturNadolski, 2021-04-26:
On two latest version of Volumio installation from source fail due to dependencies problem.

antiprism, 2021-04-26:
Hi ArturNadolski

Thank you for reporting the issue. Could you give some more details please. It would be helpful if you could copy and paste the error message.

Adrian.

ArturNadolski, 2021-04-26:
Right now I can't but it's glibc or gcc issue.
It fail on installation of packages needed to build and run cava and mpd_oled.

antiprism, 2021-04-26:
Hi ArturNadolski

Thanks. I have made a fresh install of Volumio and can see the error
```
volumio@volumio15:~$ sudo apt install build-essential                          
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 build-essential : Depends: gcc (>= 4:4.9.1) but it is not going to be installed
                   Depends: g++ (>= 4:4.9.1) but it is not going to be installed
                   Depends: dpkg-dev (>= 1.17.11) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

I'll report it to Volumio. I'll also see if the appropriate packages can be installed individually.

In the meantime, I have just checked and the binary package install is working fine

   https://github.com/antiprism/mpd_oled/blob/master/doc/install_volumio2_deb.md

Adrian.

ArturNadolski, 2021-04-26:
Thanks for the info.
Does binary is compiled from the latest version of code?

antiprism, 2021-04-26:
Yes. The current binary package was made at the latest commit https://github.com/antiprism/mpd_oled/commit/f2cf8a8.

Adrian.

ArturNadolski, 2021-04-26:
OK, good to know.

antiprism, 2021-04-26:
Report sent to Volumio forum: https://community.volumio.org/t/cannot-install-build-essential-package/46856/3

antiprism, 2022-11-05:
Closing the issue as Volumio 2 is no longer supported.

closed 2022-11-05 by antiprism