antiprism/mpd_oled #67: MoOde - ALSA loopback - SoX

FdeAlexa, 2021-05-08:
Hello Adrian,
After several tests I am sorry to inform you that the version "mpd_oled_test" has not solved the service start.
I can say that I have two different beaviours:
1. without any DAC board (device hw:2,0), using the Rpi headphone output: if the mpd_oled is started as a service it does not start with an MPD error. Stopping the service, starting play music and then starting mpd_oled manually everythings is ok until next reboot.
2. with a DAC board connected at GPIO bus (device hw:1,0): if the mpd_oled is started as a service the output is always at 44.1k. Stopping the service, starting play music and then starting mpd_oled manually everythings is ok until next reboot.
Plaese let me know how can help (with additional tests and/or additional informations) to try to solve this issue.
Regards, 
Francesco
P.S.: I have used  moOde 7.2.1.

antiprism, 2021-05-08:
Hi Francesco

Number 2. sounds strange, as in both cases the loopback device should not be opened by cava until after the playback device output has been started and the audio settings decided.

I have incorporated the changes into mpd_oled now. To be sure that the mpd_oled service is running the version with the changes could you completely uninstall mpd_oled, and then reinstall (package or source) (check `ls /usr/local/bin/mpd_oled*` and `ls /usr/bin/mpd_oled*` after removal to be sure that no mpd_oled components remain, and then reinstall).

I'll check the error with 1. How are you determining the 44.1K with 2?

Adrian.

FdeAlexa, 2021-05-08:
Hi Adrian,
About 2:. the error is:
MPD error - Failed to open "ALSA Default" (alsa); Error opening ALSA device "_audioout"; snd_pcm_hw_params_set_buffer_time_near() failed: Invalid argument.
please note that I have tested "reboot" and "shutdown and restart" both with (in the .service) with plughw:loopback and hw:loopback.

About 1; 
The DAC board is Audiophonics (9038) that has an own OLED that show this information.

Tomorrow I will do the test with new mpd_oled: sorry I am very tired: now it is 00:30 in Italy.
Regards and thank you very much,
Francesco

antiprism, 2021-05-09:
Hi Francesco

I'll wait for you to try the current mpd_oled before checking the issue using the headphone jack (as I don't have instant access to a  Pi with a headphone jack).

Regarding the playback audio rate, I am not sure how I can test that here. If the current mpd_oled doesn't solve the issue then the first steps will be to add a large delay after mpd_oled sees that Moode is playing and before cava is started. This will let you check that cava isn't running, and therefore that it doesn't have the loopback device open, and therefore that mpd_oled/cava isn't using ALSA at that time, and so any audio rate is produced by something else in the ALSA pipeline. Then, see what happens to the rate after cava starts (and opens the loopback). In the meantime, I'll see if I can check here whether Moode can register it is playing before audio playback starts, which would cause the fix in the current mpd_oled to fail.

Adrian.

antiprism, 2021-05-09:
Hi Francesco

I can see that sometimes the mpd_oled play screen appears at least a second before I can hear audio, so I think that this may be the cause of the issue. I'll ask about currentsong.txt on the  Moode forum. A possible fix would be a option for mpd_oled to delay starting cava.

Adrian.

antiprism, 2021-05-09:
Hi Francesco

I don't think the current mpd_oled will help with the issue, and so I don't think it is worth installing it. I'll add in an option first for a delay when cava starts, for you to experiment with, and if a delay solves the issue (probably both issues) I'll raise it on the Moode forum.

Adrian.

antiprism, 2021-05-09:
Hi Francesco

If you build the current mpd_oled from source (no package yet) I have added the -Z option to specify a number of seconds to delay starting cava after the first play. The default is 2 seconds, which seems like it might be enough (on my system) for music to be playing. However, the default could be increased, as it is only used the first time that music is played. 

Adrian.

FdeAlexa, 2021-05-09:
Hi Adrian,
I installed it from source and tested with different values, but I'm sorry to inform you that the problem persists: it seems that the delay is from the beginning of the mpd and not from the play. In fact, if I set the parameter -Z to 30 I have a lot of time to start the player and everything is ok; if, on the other hand, I wait for more than 30 seconds, the content of the display changes and after playback start the problem arises.
Thanks and best regards,
Francesco

antiprism, 2021-05-10:
Hi Francesco

The delay is set to start at first play after mpd_oled is started, and I believe this is working correctly.

I believe the reason for what you are seeing is that mpd_oled sometimes (not always) registers first play during mpd startup (you might see when the display sceen comes on it sometimes shows a blank play screen before changinging tp the clock stop screen). It is not straightforward to determine when Moode is playing music, but the currentsong.txt said the state was "stop" when this was occuring, so I should be able to fix it.

Adrian.

antiprism, 2021-05-10:
Hi Francesco

I have pushed an update that should stop mpd_oled interpreting the mpd startup as 'play'. This means that cava should not be started until you actually choose to play something. A delay is still needed before cava is started and the default for -Z is still 2 seconds.

It is possible that an audio source can play without using the configured Loopback device (e.g. using Moode as a bluetooth speaker: http://moodeaudio.org/forum/showthread.php?tid=3788&pid=33487#pid33487), and if the first play is like this then you will probably not get the higher sample rate when you later play from a different source.

Adrian.

FdeAlexa, 2021-05-10:
Hi Adrian,
still the same problem.
Sorry.
Thanks and best regards,
Francesco

antiprism, 2021-05-11:
Hi Francesco

I have been rebooting a lot and saw an instance of cava starting early. The reason was that the Moode currentsong.txt had not been written although the flag for the Moode startup finish had been set. mpd_oled handles this by assuming that if the currentsong.txt file is empty or missing then the user wants to take values from MPD rather than Moode, which led to the interpretation of a play state. It is now possible to specify using MPD for values with an mpd_oled option, and so I can make Moode depend on currentsong.txt and handle the empty file case. I'll push a fix for this later.

Adrian.

antiprism, 2021-05-11:
HI Francesco

You will need to build and run the latest update for the /tmp/currentsong_before_sleep.txt, and /tmp/currentsong_after_sleep.txt files to be written. The first file will say what triggered the first play, and the second will confirm that the play was still current when cava was started (assuming the fles are valid).

Adrian

FdeAlexa, 2021-05-11:
Hi Adrian,
I have installed 1 hour ago "from source" as expected. 
Have I to install directly from the binary?
Thanks and regards,
Francesco

antiprism, 2021-05-11:
Hi Francesco

You need to build from source, and also install (`sudo make install;-strip`) to use the newly built binary next time the service is restarted, and then restart the service (or reboot) .

Adrian.

antiprism, 2021-05-12:
Hi Francesco

Did you get the latest code? In the mpd_oled directory run
```
git log
```
The first entry should say
```
commit 50adae6df5d17232df2c5994e723b9ce2ec66fb6 (HEAD -> master, origin/master, origin/HEAD)
Author: Adrian Rossiter <adrian@antiprism.com>
Date:   Tue May 11 19:23:46 2021 +0200

    Moode: currentsong.txt always used with 'moode'
```
If it does not then to get the latest changes run
```
git pull
```
(and try `git log` again). Then, it should be enough to run
```
make
sudo make install-strip
```
If things don't work out, just make a fresh clone of the mpd_oled repository and build and install it (there is no need to build cava again)

Adrian.

FdeAlexa, 2021-05-12:
Great Adrian,
the version was not correct, following your procedure was ok, then I have made a fresh installation from scratch and everithing seems ok, I have tried to change the SoX parameter from 192K to 384K: it does not change immediatly but, after a reboot, everithing seems ok, again.
Of course we are talking about the configuration with the I2S DAC Board; I will test on the RpI headphone configuration and I will inform you.
Many thanks!!!
[image]

FdeAlexa, 2021-05-12:
Hello Adrian,
another good new: it is correctly running on the RpI headphone configuration too.
[image]
It seems that the error was generated by the use of loopback. I have bypassed it using the SoX with the following paramenters: 
Bit depth           16
Sample rate       96 kHz.
Thank you very much again and best regards

antiprism, 2021-05-12:
Hi Francesco

Thank you for posting your results. Does it appear then that the issue is fixed with this mpd_oled update (i.e. no further changes are needed, and the headphone error can be corrected with Moode settings)? Did you have chance to experiment with the -Z option for an idea of whether 2 seconds is a reasonable default (short, and the audio always starts within this time)?

Adrian.

FdeAlexa, 2021-05-12:
Hi Adrian, 
I think that everything is ok and the delay (-Z 2) is correct.
So: I do not have any other comment but: VERY GOOD JOB!!!
Thank you very much and best regards,
Francesco
P.S.: please let me know if and when it will be available as binary version.

antiprism, 2021-05-12:
Hi Francesco

Great. I'll close the issue then.

I have just made a binary release. I have left the first-play file reports and the -Z option in for now, if needed for debugging, but removed the mention of -Z from the help. Once it is clear that everything is working well I will remove the file reports and the -Z option and hard-code the delay.

Thanks for reporting the issue and for all your help to fix it!

Adrian.

closed 2021-05-12 by antiprism