antiprism/mpd_oled #68: hello how it use with mic without player

girs1982, 2021-05-10:
it possible? and how it run, i need to see eq when i speack to mic in raspberry pi zero

antiprism, 2021-05-10:
Hi girs1982

The mpd_oled program is written to work with MPD, so perhaps you could install MPD and then play audio from your microphone through it

   https://www.musicpd.org/doc/html/plugins.html#input-plugins

Adrian.

girs1982, 2021-05-10:
thnx i try it

girs1982, 2021-05-10:
> Вы можете легко изменить конфигурацию mpd_oled для использования микрофонного входа. Фактически, я использую микрофонный вход вместо обратной петли, потому что я мог позволить ему работать.
> 
> На моей дешевой звуковой карте USB я использовал эту команду, чтобы использовать микрофонный вход `Mpd_oled -c 'alsa,hw:5,0'`

не работает ,только часы показывает

antiprism, 2021-05-10:
Hi girs1982

> does not work, only the clock shows

The spectrum screen is only shown when the player (mpd) is playing. If you want to avoid the using the player then you will need to change the mpd_oled source code (but I cannot help with this).

Adrian.

girs1982, 2021-05-11:
> Привет девочки1982
> 
> > не работает, только часы показывают
> 
> Экран спектра отображается только при воспроизведении плеера (mpd). Если вы хотите избежать использования плеера, вам нужно будет изменить исходный код mpd_oled (но я не могу с этим помочь).
> 
> Адриан.

with mpd all ok, thnx , but Mpd_oled -c 'alsa,hw:5,0' not work

supercrab, 2021-05-11:
You might need to replace `hw:5,0` with the name of your USB device, you should be able to find it using `aplay -l` and will need to test all the device names and see which one works.
You will still need to change `mpd_oled` source to always display the spectrum analyser.

girs1982, 2021-05-11:
> Возможно, вам придется заменить `hw:5,0`имя вашего USB-устройства, вы сможете найти его с помощью, `aplay -l`и вам нужно будет проверить все имена устройств и посмотреть, какое из них работает.
> Вам все равно придется менять `mpd_oled`источник, чтобы анализатор спектра всегда отображался.

i done it but see nothing, i use i2s mic hw:1,0

supercrab, 2021-05-11:
What happens if you do `arecord -D hw:1,0 -V stereo -r 44100 -f S16_LE -c 1 /dev/null`?
Does the mic work with mpd_oled when mpd is playing?

girs1982, 2021-05-11:
> Что произойдет, если вы это сделаете `arecord -D hw:1,0 -V stereo -r 44100 -f S16_LE -c 1 /dev/null`?
> Работает ли микрофон с mpd_oled при воспроизведении mpd?

wuth mpd all work , i try  it arecord -D hw:1,0 -V stereo -r 44100 -f S16_LE -c 1 /dev/null

supercrab, 2021-05-11:
What happens when you run `arecord -D hw:1,0 -V stereo -r 44100 -f S16_LE -c 1 /dev/null `?

supercrab, 2021-05-11:
You might have to stop the `mpd_oled` process with `killall mpd_oled`?

girs1982, 2021-05-11:
> это необходимо для проверки того, что вы используете правильное имя аппаратного устройства и правильно ли получаете сигнал.

it will work after reboot, what command need to start mic & mpd_oled?

girs1982, 2021-05-11:
> Я думал, что так будет, `sudo mpd_oled -o 1 -c 'alsa,hw:1,0'`но ты сказал, что это не сработало?

Error loading config. cava was built without 'alsa' input support

girs1982, 2021-05-11:
no it error was then mpd player was run, i kill it, an erros no, but in oled only clock

supercrab, 2021-05-11:
I think you need to build CAVA with ALSA support but I don't know how to do that :(

girs1982, 2021-05-11:
> Я думаю, вам нужно создать CAVA с поддержкой ALSA, но я не знаю, как это сделать :(

i make cava 
git clone https://github.com/karlstav/cava
cd cava
./autogen.sh
./configure --disable-input-portaudio --disable-input-sndio --disable-output-ncurses --disable-input-pulse --program-prefix=mpd_oled_
make
sudo make install-strip

girs1982, 2021-05-11:
maybe how you done it?

supercrab, 2021-05-11:
Maybe you could try it from my fork?

https://github.com/supercrab/mpd_oled

girs1982, 2021-05-11:
> Maybe you could try downloading and building `mpd_oled` from my fork?
> 
> https://github.com/supercrab/mpd_oled

i rebuild cava but not work, with mpd all work

supercrab, 2021-05-11:
What do you mean it doesn't work?

girs1982, 2021-05-11:
> What do you mean it doesn't work?

wtiout mpd player

antiprism, 2021-05-11:
Hi girs1982

By default, cava will include the functionality available when it is built, so if an ALSA development package isn't installed then cava will not include ALSA support.

If you are using a Debian-based OS that doesn't include an ALSA development package then you can install it with

   sudo apt install libasound2-dev

If you then clone a fresh copy of cava and build it, the cava binary should include ALSA support (check the log when you run `configure`).

Adrian.

girs1982, 2021-05-12:
> Hi girs1982
> 
> The mpd_oled program is written to work with MPD, so perhaps you could install MPD and then play audio from your microphone through it
> 
> https://www.musicpd.org/doc/html/plugins.html#input-plugins
> 
> Adrian.

how it compile with wiringPi?
error: conflicting declaration of C function ‘void bcm2835_delayMicroseconds(unsigned int)’

antiprism, 2021-05-12:
Hi girs1982

Sorry, I can't help with programming questions,

Adrian.

antiprism, 2021-07-22:
No further replies, so closing.

closed 2021-07-22 by antiprism