antiprism/mpd_oled #70: Mod main.cpp

fabio323ti, 2021-07-22:
I modded main.cpp to extend to full screen spectrum bar , edit bitrate posizion, volume slide width .. very nice.
I wish to edit the height of spectrum to avoid it goes over top bitrate..
i attach my main.cpp modified.

and if someone can help me with strange Oled 128x64 issue  ..it often switch up some lines, already tryed to down dtparam to 40000 without succes... maybe some tricks?

with binary volumio install of plugin this doesn't happen

Many thanks

antiprism, 2021-07-22:
Hi

The binary package of mpd_oled is made from the current commit. If you are using a modified main.cpp from an older commit then maybe you are missing the line
```
display.reset_offset();
```
Adrian.

fabio323ti, 2021-07-22:
in main.cpp ther's not display.reset_offset(); line.
adding at end of file give error on compiling

antiprism, 2021-07-22:
Hi

Here is the line

https://github.com/antiprism/mpd_oled/blob/0e64706386cdee8ecd43013a834f893e87e1c803/src/main.cpp#L607

You may need to make some other changes too (you could check the mpd_oled commit log).

I am closing the issue as it is fixed in the latest mpd_oled.

Adrian.

closed 2021-07-22 by antiprism