antiprism/mpd_oled #71: In my case latest release not solve.

fabio323ti, 2021-07-22:
Hi

Here is the line

https://github.com/antiprism/mpd_oled/blob/0e64706386cdee8ecd43013a834f893e87e1c803/src/main.cpp#L607

You may need to make some other changes too (you could check the mpd_oled commit log).

I am closing the issue as it is fixed in the latest mpd_oled.

Adrian.

_Originally posted by @antiprism in https://github.com/antiprism/mpd_oled/issues/70#issuecomment-885156922_

antiprism, 2021-07-22:
Hi

You posted that the Volumio plugin binary install of mpd_oled was working for you

   https://github.com/antiprism/mpd_oled/issues/70#issue-950629231

But this installs the latest mpd_oled binary package that I make

  https://github.com/supercrab/volumio-plugins/blob/master/plugins/miscellanea/mpd_oled/install.sh

Suggesting that the latest version of mpd_oled does not have the issue you are seeing in your customised version.

Adrian.

fabio323ti, 2021-07-22:
Thanks you Adrian, i will use binary version without any skin mods..
if u consider to share some mods about skin feature, very appreciated.

Best will be full large 128px spectrum without time o'clock. nothing else.
thanks

antiprism, 2021-07-23:
Hi

Thanks for confirming that the latest version of mpd_oled is working for you.

I am in the process of making some major changes to mpd_oled, and these include adding some extra screen layouts, and so at some stage in the not-too-distant future (but I cannot say when) mpd_oled will include an option for a spectrum-only display.

Closing the issue.

Adrian.

closed 2021-07-23 by antiprism