antiprism/mpd_oled #72: Feature Request: Turn off display or screensaver after idle period

shamasis, 2021-07-30:
First off, I am a huge fan of the work done for Volumio and now the mod_oled project. I understand the effort and love that goes into building something like this.

I would love to see a setting that allows the display to be turned off after a period of mid/volumio inactivity.

If I can figure it put, I will myself try and send a PR (but it seems too far off from my ability to get there.)

antiprism, 2021-07-30:
Hi

Thank you for your suggestion to add a screensaver. This has been previously been requested and I do intend to add it in (in my own case continually displaying the stop screen has led to screen burn).

I will be adding this to the development code

   https://github.com/antiprism/mpd_oled_dev

I'll leave the issue open until the feature is available (but I cannot give a time frame for this).

Adrian.

petermolnar, 2023-08-10:
I needed a quick and dirty solution for this until I have some time to get familiar with your code (I only added the display today), so I wrote a loop in bash that starts/stops the service depending if mpd is playing:

`/usr/local/bin/mpd_oled_cron`

```bash

#!/bin/bash

cntr=0

while /bin/true; do
  if mpc | grep -q playing; then
    cntr=0
    if [[ $(pgrep -c mpd_oled) -eq 0 ]] ; then
      systemctl start mpd_oled
    fi
  else
    if [[ $(pgrep -c mpd_oled) -gt 0 ]] ; then
      cntr=$((cntr+1))
    fi

    if [[ $cntr -gt 30 ]]; then
      systemctl stop mpd_oled
    fi
  fi
  sleep 2
done
```

`/lib/systemd/system/mpd_oled_cron.service`
```
[Unit]
Description=Starts/stops mpd_oled

[Service]
Type=simple
ExecStart=/bin/bash /usr/local/bin/mpd_oled_cron
Restart=always

[Install]
WantedBy=multi-user.target
```

zsoltmar, 2023-09-01:
I would love to have a ping-pong animation going on after say 10 minute of inactivity. This would use all pixels and make the time readout undisturbed (which is pretty much the main function if not playing anything). This could work in conjunction with the screen invert timer.

shamasis, 2024-07-30:
Hi @antiprism

After about 2 years of continuous run, I have gone through 3 displays. Keeping them on burns the clock image and after a while due to the burn-in, even normal scrolling texts are hard to read.

antiprism, 2024-07-30:
Hi shamasis

I appreciate that burn-in is an issue, but I am no longer making any changes to this repository except for bugs that stop mpd_oled running (and only then if they are easy to fix).

Adrian.